CadQuery/cadquery #1329: ERROR: No matching distribution found for cadquery-ocp<7.8,>=7.7.0a0 (from cadquery)

bzuidgeest, 2023-05-02:
When trying to install cadquery using pip I get the result below. I am no python expert, I have seen similar questions, but could not see how to solve this.
I got this on my linux mint install and also ubuntu under wsl2.

python version is 3.8.10
pip version is 20.0.2
platform x86-64

 pip install cadquery
Collecting cadquery
  Downloading cadquery-2.2.0-py3-none-any.whl (142 kB)
     |████████████████████████████████| 142 kB 11.2 MB/s
Collecting ezdxf
  Downloading ezdxf-1.0.3-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_12_x86_64.manylinux2010_x86_64.whl (3.2 MB)
     |████████████████████████████████| 3.2 MB 20.2 MB/s
Collecting nlopt
  Downloading nlopt-2.7.1-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (424 kB)
     |████████████████████████████████| 424 kB 61.2 MB/s
Collecting typish
  Downloading typish-1.9.3-py3-none-any.whl (45 kB)
     |████████████████████████████████| 45 kB 5.0 MB/s
Collecting multimethod<2.0,>=1.7
  Downloading multimethod-1.9.1-py3-none-any.whl (10 kB)
Collecting casadi
  Downloading casadi-3.6.2-cp38-none-manylinux2014_x86_64.whl (67.5 MB)
     |████████████████████████████████| 67.5 MB 1.3 MB/s
Collecting path
  Downloading path-16.6.0-py3-none-any.whl (26 kB)
**ERROR: Could not find a version that satisfies the requirement cadquery-ocp<7.8,>=7.7.0a0 (from cadquery) (from versions: none)
ERROR: No matching distribution found for cadquery-ocp<7.8,>=7.7.0a0 (from cadquery)**

bzuidgeest, 2023-05-02:
Ok I figured out that is seems those packages only support windows. I also found that for any recent build of cq-editor I had to move to a fork.
https://github.com/jmwright/CQ-editor
as the master only has builds that are very old.

With all due respect to the project owners for providing this all for free. Setting this up is not as easy as it should be and that means a project might not get the attention it deserves.

jmwright, 2023-05-02:
Older versions of pip (like your version) will show this error. That's why the first step in the installation instructions is to upgrade pip.

https://github.com/CadQuery/cadquery#cadquery-installation-via-pip

pip installation supports Windows, Linux and MacOS, not just Windows.

bzuidgeest, 2023-05-02:
Thank you for responding.

I tried upgrading pip as stated. Just didn't change anything... I know from what I have read that pip is available on all those platforms. But I read that something called as "wheel" can be restricted to a certain platform. And that the error I get is one you get when your platform is not in the "wheel". As I said, not a python guy.

after a pip install pip --upgrade I am now on version 23.1.2

charlylima, 2023-12-16:
If "pip install cadquery" does not work, try pip3 command, e.g. on Ubuntu Linux 20.04:

```
# Install python library
sudo apt install python3 python3-pip
pip3 install cadquery

# Install graphical editor
pip3 install PyQt5 spyder pyqtgraph logbook
pip3 install git+https://github.com/CadQuery/CQ-editor.git

# Run
cq-editor
```