CadQuery/cadquery #1333: Dynamic addresses?

Redsparkfish, 2023-05-15:
I am trying to build a face-edge graph from a model in STP form. Just as its name, the vertices of the graph are the faces of the model and the edge between 2 vertices are the edge between 2 adjacent faces. So I need to iterate over faces and edges of the model. 

However, it seems that the addresses of the workplane objects are dynamic. When I print model.faces().all() for several times, the results are all different. Also, when I take an edge from a face, the edge is not likely to be in model.edges.all().

jmwright, 2023-05-15:
I believe this is the way the underlying CAD kernel works. We can't always guarantee IDs will be the same, and the order in which objects are returned can vary. I've experimented with tagging features with hashes of their properties (length, area, volume, etc), and have used that as a way to track them. Of course these hashes change as you add or change those features (holes, extrusions, etc) so it's only useful in some cases.

adam-urbanczyk, 2023-05-18:
I don't think cadquery will help you with that, you'll need to make yourself familiar with the underlying cad kernel (OCCT). There are chances, that what you need is already implemented.