CadQuery/cadquery #1335: Update the TJS export format

agualdron, 2023-05-19:
## Issue

The format that uses the TJS exporter is no longer supported by Three.js. 

### Request

Update the TJS exporter to align with the currently supported format. Refer to the [JSON Object Scene format 4 documentation](https://github.com/mrdoob/three.js/wiki/JSON-Object-Scene-format-4). 

This update would allow users to load 3D scenes into Three.js in an online environment.

adam-urbanczyk, 2023-05-21:
How about adding GLTF support to non-assy exporters instead?

agualdron, 2023-05-24:
that would be better instead!
currently, is there a way to export to GLTF?

adam-urbanczyk, 2023-05-25:
Yes, via `cq.Assembly`, something like this should work:
```python
cq.Assembly().add(obj).save('filename.gltf')
```