CadQuery/cadquery #1337: How to loft between 2 defined wires?

geeklimit, 2023-05-23:
I'm very new to cadquery, but I've managed to get the ends of an object defined how I'd like them.

I'm trying to now loft between the two wires, then round all edges. I can't seem to figure out how to loft, and any help is appreciated.

```
import cadquery as cq
result0 = (cq.Workplane("XY")
           .moveTo(0, 0)
           .lineTo(30, 0)
           .lineTo(30,25)
           .lineTo(20,25)
           .lineTo(20,20)
           .lineTo(25,20)
           .lineTo(25,5)
           .lineTo(0,5)
           .mirrorY()
           )

result1 = (cq.Workplane("XY")
           .workplane(offset=60)
           .moveTo(0, 0)
           .lineTo(30, 0)
           .lineTo(30,25)
           .lineTo(20,25)
           .lineTo(20,20)
           .lineTo(25,20)
           .lineTo(25,5)
           .lineTo(0,5)
           .mirrorY()
           )
result1 = result1.rotate((0, 0, 0), (1, 0, 0), 3.865)
```

lorenzncode, 2023-05-23:
Given that you have two separate Workplane objects result0 and result1, you might use the direct API as follows to create the loft:

```py

...

result = cq.Solid.makeLoft([result0.wires().val(), result1.wires().val()])
show_object(result)
```

To use the Workplane `loft` method (as an alternative solution), the set of wires must be defined in the same Workplane object.

```
import cadquery as cq

result0 = (
    cq.Workplane("XY")
    .moveTo(0, 0)
    .lineTo(30, 0)
    .lineTo(30, 25)
    .lineTo(20, 25)
    .lineTo(20, 20)
    .lineTo(25, 20)
    .lineTo(25, 5)
    .lineTo(0, 5)
    .mirrorY()
)

result = (
    result0.translate((0, 0, 60)).rotate((0, 0, 0), (1, 0, 0), 3.865).toPending().loft()
)
show_object(result)
```

Here _result_ builds on _result0_ and the first wire is already part of the modeling context "pendingWires" due to the `mirrorY` call.  Then you must add the explicit `toPending` call to add the second wire and make it available for the pending `loft` operation.

geeklimit, 2023-05-24:
@lorenzncode How would I loft on a curve? This is a straight loft, and I'm trying to fit this channel onto a surface with a 40-inch radius.

lorenzncode, 2023-05-24:
> I'm trying to fit this channel onto a surface with a 40-inch radius.

Try revolve.

```py
import cadquery as cq

result0 = (
    cq.Workplane("XY")
    .lineTo(30, 0)
    .lineTo(30, 25)
    .lineTo(20, 25)
    .lineTo(20, 20)
    .lineTo(25, 20)
    .lineTo(25, 5)
    .lineTo(0, 5)
    .mirrorY()
)

result = result0.revolve(90, (0, -40, 0), (1, -40, 0))
```

geeklimit, 2023-06-02:
Thank you so much. This is what I needed. The final script ended up being:

```
import cadquery as cq

#channel dimensions
length = 60
tablength = 10
width = 60
height = 10
thickness = 6
degrees = (length / 10) * 0.64423 #for a 40in (1016mm) radius, 10mm of length = 0.644 degrees
pipeOD = 21.336

result0 = (
    cq.Workplane("XY")
    .moveTo(0,0)
    .threePointArc((10.668,-10.668),(7.555,-10.668 - 7.555))
    .lineTo(7.555 + thickness, -10.668 - 7.555)
    .threePointArc((10.668 + thickness,-10.668),(thickness * 2,0))
    
    .lineTo((width / 2) + thickness, 0)    
    .lineTo((width / 2) + thickness, height + (thickness * 2))
    .lineTo((width / 2) - tablength, height + (thickness * 2))
    .lineTo((width / 2) - tablength, height + thickness)
    .lineTo((width / 2), height + thickness)
    .lineTo((width / 2), thickness)
    .lineTo(0, thickness)
    .mirrorY()
)

result = (result0.revolve(degrees, (0, -1016, 0), (1, -1016, 0)).fillet(thickness / 2.5)
)
```

closed 2023-06-02 by geeklimit