CadQuery/cadquery #1338: Issue creating STEP file after revolving arc

tney314, 2023-05-30:
I'm new to CadQuery and have run into a repeated issue creating STEP files while trying to revolve arcs. It appears to be an issue with the export function itself and possibly OpenCascade, but I was wondering if anyone has experienced anything similar or sees something I am doing wrong!

I've tried both methods of creating a sketch and revolving it as one solid body, as well as creating a body and then revolving a sketch to cut it. Both have caused the same error. The revolve -> step conversion works fine for anything from 0-287 degrees, but as soon as I pass 288-360 the error occurs (see images), and has done so consistently on multiple different projects.

Here is a simple code I wrote to test for errors after I kept seeing it come up elsewhere:
import cadquery as cq
result = (cq.Workplane("front")
          .lineTo(2.0, 0)
          .threePointArc((1.0, 1.0), (0.0, 0.0))
          .close()
          .revolve(360, (0, 0), (0, 1))
    )
cq.exporters.export(result, 'C:Path/result.step')

Here are some images after opening the STEP files in FreeCAD. Each worked as expected in the native CQ-editor, but created problems when opening in other apps.
Rotating 287 degrees: Note that this worked as expected
[image]

Rotating 288 degrees: Note that this created what seems like the inverse of the revolve?
[image]

Rotating 360 degrees: Note that this created a full revolve, but also created a torus not as expected
[image]

adam-urbanczyk, 2023-05-30:
Maybe try making the object not degenerate:

```python
import cadquery as cq

eps=1e-3

result = (cq.Workplane("front")
.moveTo(eps, 0)
.lineTo(2.0, 0)
.threePointArc((1.0, 1.0), (eps, 0.0))
.close()
.revolve(360, (0, 0), (0, 1))
)
```

adam-urbanczyk, 2023-05-31:
The code above removes the degenerate point in the middle (zero length edge). Failures of the kernel are often related to this kind of situations.