CadQuery/cadquery #1339: Should we mention the CQ-editor installer in the README?

adam-urbanczyk, 2023-06-03:
@jmwright  @lorenzncode See https://github.com/CadQuery/CQ-editor/releases/tag/nightly.

lorenzncode, 2023-06-03:
Very nice - this was easy to use and great for someone unfamilar with conda!  And it allows you to install extra packages if you need to.

Yes I would propose to mention in the README and installation instructions.
1. documentation: Detailed instructions in the Installation section
2. README: mention the CQ-editor installer prominently and link to docs

I tried the installer on Linux and Windows and created a PR #1340  with doc updates - if you have something else in mind I can comment on a separate PR instead.

adam-urbanczyk, 2023-06-04:
Closed with #1340

closed 2023-06-04 by adam-urbanczyk