CadQuery/cadquery #1341: Lofting with guide curves

rotofyne, 2023-06-05:
Hello, is it possible to create a loft between multiple section and have guide curves between sections to constrain the cross section of the loft between the specified sections? #1337 seems to imply that may not be possible however I wanted to confirm if that was the case.

Jojain, 2023-06-05:
I think I had been quite clear in the other issue. No you cannot. What you can do is use multi_sweep. It takes several sections and a spine. I.e a curve passing through the center of the cross sections

rotofyne, 2023-06-05:
@Jojain thanks, I assumed that's what you meant in the other issue but I hadn't seen the phrase "spine curves" before so wanted to confirm it was the same as "guide curves". It looks like sweep_multi only takes a single wire from the class summary which would not be helpful for my use case.  @adam-urbanczyk it looks like the "Gordon Surface" you mentioned in issue #497 would work. I'd be interested in doing a PR to try to port over curve_network_to_surface. Have you looked at this at all recently?

closed 2023-06-29 by rotofyne