CadQuery/cadquery #1345: Creating multiple objects in one STL file

tobias-mierzwa, 2023-06-12:
Hello together,

I have a question about exporting multiple objects in a stl file. What i want is, when I import the exported stl-file into the Cura software, the objects should be listed individually, and when I slice it, each object should be finished one at a time, instead of adding a layer to all objects in the same iteration in the print.

my current snippet for creating simple cylinder objects

```python
for point in self.__create_array_points(cylinder_radius * 2, distance, quantity):
            workplane = Workplane()
            workplane.pushPoints([point]).cylinder(cylinder_height, cylinder_radius)
```

The following issue may address the same problem https://github.com/CadQuery/cadquery/issues/212, but I was not able to achieve the result I wanted due to my lack of knowledge.

adam-urbanczyk, 2023-06-12:
Multiple objects can be exported in one STL, but AFAIK STL does not convey topology information. Maybe try 3MF? And what is your question actually?

tobias-mierzwa, 2023-06-13:
Sorry if I did not make myself clear. So my current code to generate multiple objects can be seen in the following snippet:

```python
def _cylinder_set(
        self,
        cylinder_height: float,
        cylinder_radius: float,
        quantity: int,
        distance: int = 1,
    ) -> Workplane:
        return self.workplane.pushPoints(
            self.__create_array_points(cylinder_radius * 2, distance, quantity)
        ).cylinder(cylinder_height, cylinder_radius)
```

This snippet results in the following STL when I create 5 cylinders and is shown in the next screenshot.

[image]

You can see in the screenshot that all 5 cylinders are counted as one object, so the five cylinders get one layer each iteration.

What I really want to do can be seen in the next screenshot, where two cylinders are listed individually in the object list and are printed one after the other. So the printer finishes one cylinder before moving the head to the next cylinder position.

[image]

I hope I have been able to make my problem more understandable with the information I have provided

adam-urbanczyk, 2023-06-13:
Long story short, use `3mf` instead of `stl`.

It is not clear that `stl` officially supports multiple solids (see http://www.fabbers.com/tech/STL_Format ).

tobias-mierzwa, 2023-06-13:
Thanks for the hint!

Still gives me the problem of how I export the individual cylinders in one file (3mf)

The following snippet is not providing the solution I want. Is still one object in Cura. 

```python
        workplanes = []

        for point in self.__create_array_points(cylinder_radius * 2, distance, quantity):
            workplane = Workplane()
            workplane = workplane.pushPoints([point]).cylinder(cylinder_height, cylinder_radius)
            workplanes.append(workplane.val())

        exporters.export(Compound.makeCompound(workplanes), "test.3mf")
 ```
 
 
[image]

I am not sure if the solution is to export multiple compound objects in a 3mf file? But the exporter API does not provide a function to achieve this.

adam-urbanczyk, 2023-06-18:
@tobias-mierzwa could you check if #1351 behaves as you expect?

tobias-mierzwa, 2023-06-19:
> @tobias-mierzwa could you check if #1351 behaves as you expect?

Unfortunately not. I am very happy to provide more information or files if it helps.
But if the use case is too specific, we can also close the issue for now.

adam-urbanczyk, 2023-06-20:
Could you check with Cura devs how the 3mf file needs to be structured to achieve what you want?

Scaremonger, 2026-11-03:
I know this is an old post, but maybe related to an issue I am experiencing.
Running the above example and opening test.3mf in Orcaslicer shows "Error: 1518 Non-Manifold Errors".
My stripped back code to identify the problem is below and this gives "24 Non-Manifold Errors" in the two 3mf files and none in the stl:
```
#!/usr/bin/env python3
import cadquery as cq
from cadquery import exporters, Compound

if __name__ == "__main__":
    # Create a model
    model = cq.Workplane().box(10, 10, 10)

    # Save as an STL
    cq.exporters.export( model, "example.stl" )

    # Save as a 3MF
    compound = Compound.makeCompound( model )
    exporters.export( compound, "example1.3mf")
    
    model.export( "example2.3mf", tolerance=0.01, angularTolerance=0.1)
```
I am running Python 3.10.12 on Linux Mint 21 with cadquery==2.7.0 and cadquery-ocp==7.8.1.1.post1. Orcaslicer v2.4.2 build 8500fcdc
Any suggestions?