CadQuery/cadquery #1346: GMSH integration

adam-urbanczyk, 2023-06-13:
With OCP 7.7.1 it should be possible to import OCCT shapes from memory
```python
import cq
import gmsh

gmsh.initialize()

gmsh.model.occ.importShapesNativePointer(cq.Workplane().box(1,1,1).val()._address())
```

- [ ] Check if it actually works

shimwell, 2023-06-13:
Running the above (recently updated) script works over here..
[image]

shimwell, 2023-06-14:
When exporting workplanes that have surfaces in contact it is nice to merge those surfaces together. This allows the meshing to be conformal at the surface boundary between the two workplanes.

I think this code does the same as above but works for two workplanes that contact on a surface.

```python
import cadquery as cq
import OCP
import gmsh

workplane1 = cq.Workplane("XY").cylinder(height=10, radius=4)
workplane2 = cq.Workplane("XY").cylinder(height=10, radius=5).cut(workplane1)

cq.exporters.export(workplane1, "workplane1.step")
cq.exporters.export(workplane2, "workplane2.step")

bldr = OCP.BOPAlgo.BOPAlgo_Splitter()

bldr.AddArgument(workplane1.val().wrapped)
bldr.AddArgument(workplane2.val().wrapped)

bldr.SetNonDestructive(True)

bldr.Perform()

bldr.Images()

merged_surface_compound = cq.Compound(bldr.Shape())

# it looks like cq.Compound does not have a toOCC method like cq.workplane does
# however it has a val().wrapped which returns a OCP.TopoDS.TopoDS_Solid
topods = merged_surface_compound.wrapped

gmsh.initialize()
gmsh.option.setNumber("General.Terminal", 1)
volumes = gmsh.model.occ.importShapesNativePointer(topods._address())
gmsh.model.occ.synchronize()
gmsh.option.setNumber("Mesh.Algorithm", 1)
gmsh.option.setNumber("Mesh.MeshSizeMin", 20)
gmsh.option.setNumber("Mesh.MeshSizeMax", 30)
gmsh.model.mesh.generate(2)
gmsh.write("gmsh_of_cadquery_compound_in_memory.msh")
```

I shall confirm if this works or not soon

shimwell, 2023-06-14:
Confirming that the surface merge and mesh worked

As we can see from this screenshot the resulting gmsh has just 6 surfaces instead of the 7 surfaces made in cadquery because the contact surface has been merged
[image]

adam-urbanczyk, 2023-06-15:
Great! Thanks for the investigation @shimwell .

closed 2023-08-10 by adam-urbanczyk