CadQuery/cadquery #1352: Offline version of documentation (zipped HTML)

fe60, 2023-06-19:
Without or with a poor internet connection it would be helpful to have an offline version of the documentation.
Please consider generating a downloadable zipped HTML.

https://docs.readthedocs.io/en/stable/downloadable-documentation.html
https://docs.readthedocs.io/en/stable/guides/enable-offline-formats.html

juhldesign, 2023-07-19:
I am going to second this, read the docs offers a pdf export but it appears to be disabled for CadQuery

closed 2023-07-24 by jmwright

jmwright, 2023-07-24:
I'm reopening this issue even though the PR closed it. There is still more work to be done. The dynamic renders are ignored in the PDF and epub versions. The zipped HTML seems to work as expected though.

PR #1377 (2023-07-24): Making all download formats of the RTD documentation available
Closes #1352 

It should make zipped HTML, PDF and epub formats available for download.