CadQuery/cadquery #1364: FixedAxis assembly constraint VTK render is broken

sethfischer, 2023-07-04:
See https://cadquery.readthedocs.io/en/latest/assy.html#fixedaxis

The VTK render window is blank and directive `.. cadquery::` fails with:

```commandline
$ rm -r target/docs/ && ./build-docs.sh
Running Sphinx v5.3.0
⋮
updating environment: [new config] 17 added, 0 changed, 0 removed
Traceback (most recent call last):                                                                                                                                                                  
  File "...cadquery/cadquery/cqgi.py", line 564, in visit_Assign
    for n, v in zip(left_side.elts, node.value.elts):
AttributeError: 'Name' object has no attribute 'elts'
Unable to handle assignment for node 'Name(id='pnt', ctx=Store())'
reading sources... [100%] sketch                                                                                                                                                                    
⋮
build succeeded.

The HTML pages are in ../target/docs.
```

adam-urbanczyk, 2023-07-04:
Probably caused by #1353

sethfischer, 2023-07-04:
Not sure how far back it goes but the bug exists in cb1af14aed2e480f248571597f973b706b5cd8c1 which is the commit prior to #1353.

sethfischer, 2023-07-05:
Using `git bisect` I've established that the VTK rendering of this example was broken with commit d71458a195f652614e7d815a34beb4cf7a463c96.

PR #1367 (2023-07-08): Fix empty render in docs and doc build error
Closes #1364
closed 2023-07-08 by adam-urbanczyk