CadQuery/cadquery #1371: Crash on exit (Python has stopped working) Windows 10

clayton13, 2023-07-08:
This occurs in any script/program that imports cadquery upon exit. I was running some python scripts from [kicad-packages3D-generator](https://gitlab.com/kicad/libraries/kicad-packages3D-generator) and did not realize they were successful as the python crash was misleading. Affecting Cq-editor, scripts and the interpreter. 

## To Reproduce

```python
import cadquery
exit(0)
```

[image]

## Backtrace

From windows event viewer -> application logs

C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_python.exe_887f109fbd15ed3565f272f795e23c6c1fac36_240a4830_c22050e2-e26b-454a-bdc5-1347496224f9

[Report.txt](https://github.com/CadQuery/cadquery/files/11991455/Report.txt)
[app_error.txt](https://github.com/CadQuery/cadquery/files/11991459/app_error.txt)

```
Faulting application name: python.exe, version: 3.11.4150.1013, time stamp: 0x64801a28
Faulting module name: ntdll.dll, version: 10.0.19041.3086, time stamp: 0x6349a4f2
Exception code: 0xc0000005
Fault offset: 0x00000000000a6ffb
Faulting process id: 0x7598
Faulting application start time: 0x01d9b1c6b16253f9
Faulting application path: C:\Program Files\Python311\python.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 5a604d36-5944-48cc-a082-3001351c6f7d
Faulting package full name: 
Faulting package-relative application ID: 
```

## Environment

I am not using conda, so that is potentially a contributing factor. 

```
C:\Users\Clayton>systeminfo | findstr /B /C:"OS Name" /B /C:"OS Version"
OS Name:                   Microsoft Windows 10 Pro
OS Version:                10.0.19045 N/A Build 19045
```

Fresh install of Python (immediately before installing cq).

```
C:\Users\Clayton>where python
C:\Program Files\Python311\python.exe
C:\Users\Clayton>python --version
Python 3.11.4
C:\Users\Clayton>pip --version
pip 23.1.2 from C:\Program Files\Python311\Lib\site-packages\pip (python 3.11)
C:\Users\Clayton>pip list
Package           Version
----------------- -------
asttokens         2.2.1
backcall          0.2.0
cadquery          2.3.1
cadquery-ocp      7.7.0
casadi            3.6.3
colorama          0.4.6
decorator         5.1.1
executing         1.2.0
ezdxf             1.0.3
ipython           8.14.0
jedi              0.18.2
matplotlib-inline 0.1.6
multimethod       1.9.1
nlopt             2.7.1
nptyping          2.0.1
numpy             1.25.0
parso             0.8.3
path              16.7.1
pickleshare       0.7.5
pip               23.1.2
prompt-toolkit    3.0.39
pure-eval         0.2.2
Pygments          2.15.1
pyparsing         3.1.0
PyYAML            6.0
setuptools        65.5.0
six               1.16.0
stack-data        0.6.2
traitlets         5.9.0
typing_extensions 4.7.1
typish            1.9.3
wcwidth           0.2.6
```

Using: Affecting Cq-editor, scripts and the interpreter.

jmwright, 2023-07-08:
Can you try using Python 3.10? One of our dependencies (nlopt) does not have proper Python 3.11 support yet.

clayton13, 2023-07-08:
Good catch @jmwright. No crash experienced with Python 3.10.11