CadQuery/cadquery #1375: importStep ignore dimensions

HenningScheufler, 2023-07-13:
Dimensions inside the step file are not recognized:

```python
import cadquery as cq
import math

r = 1
h = 1
Vol = r**2 * math.pi * h

r_wrong = 1000
h_wrong = 1000
Vol_wrong = r_wrong**2 * math.pi * h_wrong
stp = cq.importers.importStep("cylinder_1000mm.step")
print("volume: ", stp.objects[0].Volume(), " corrected volume: ",r**2 * math.pi * h)
print("volume with wrong dimensions: ", Vol_wrong) 
```

results in

```
volume:  3141592653.5897927  corrected volume:  3.141592653589793
volume with wrong dimensions:  3141592653.589793
```

The step file was created with freecad. The same issue appeared with a 3Dx step file.

adam-urbanczyk, 2023-07-14:
Can you share the step file? NB: CQ has no units.

adam-urbanczyk, 2024-01-23:
So the defualt import/export unit of OCCT seems to be mm. You can change it for importing using `xstep.cascade.unit` via e.g. `OCP.Interface.Interface_Static.SetCVal_s("write.step.unit","UM")`