CadQuery/cadquery #1382: Error of sweep operation: OCP.Standard.Standard_RangeError: NCollection_Array1::Create

bladesaber, 2023-07-25:
## To Reproduce

```python
import cadquery as cq
import numpy as np
import math

path_xyz_tuples = [
    (0.0, 0.0, 0.0),
    (1.2, 0.0, 0.2),
    (2.4, -0.0, 1.6),
    # (2.8, -0.0, 3.5),
]
path = cq.Workplane("XY").polyline(path_xyz_tuples, includeCurrent=False)
# path = cq.Workplane("XY").spline(path_xyz_tuples)

vs = path.vertices()

r1 = cq.Workplane("YZ").circle(1.315)
tube = r1.sweep(path, makeSolid=True)
```

## Backtrace

```
Traceback (most recent call last):
  File "/home/***/Desktop/temptory/temProject/test3.py", line 17, in <module>
    tube = r1.sweep(path, makeSolid=True)
  File "/home/***/anaconda3/envs/py39/lib/python3.9/site-packages/cadquery/cq.py", line 3151, in sweep
    r = self._sweep(
  File "/home/***/anaconda3/envs/py39/lib/python3.9/site-packages/cadquery/cq.py", line 3759, in _sweep
    thisObj = Solid.sweep(f, p, makeSolid, isFrenet, mode, transition)
  File "/home/***/anaconda3/envs/py39/lib/python3.9/site-packages/cadquery/utils.py", line 47, in __call__
    return super().__call__(*args, **kwargs)
  File "/home/***/anaconda3/envs/py39/lib/python3.9/site-packages/multimethod/__init__.py", line 315, in __call__
    return func(*args, **kwargs)
  File "/home/***/anaconda3/envs/py39/lib/python3.9/site-packages/cadquery/occ_impl/shapes.py", line 3412, in sweep
    return cls.sweep(
  File "/home/***/anaconda3/envs/py39/lib/python3.9/site-packages/cadquery/utils.py", line 47, in __call__
    return super().__call__(*args, **kwargs)
  File "/home/***/anaconda3/envs/py39/lib/python3.9/site-packages/multimethod/__init__.py", line 315, in __call__
    return func(*args, **kwargs)
  File "/home/***/anaconda3/envs/py39/lib/python3.9/site-packages/cadquery/occ_impl/shapes.py", line 3389, in sweep
    builder.MakeSolid()
OCP.Standard.Standard_RangeError: NCollection_Array1::Create

Process finished with exit code 1
```

## Info
When I set the 'makeSolid' of the sweep function to False. I got a picture below:
[image]

Is it something wrong with the use of the sweep function?

Thanks and looking forward to your response as soon as possible.

## Environment

OS: Ubuntu 20.04

Was CadQuery installed using Conda?: 
Output of `conda list` from your active Conda environment:
```
python     3.9.17
cadquery 2.3.1  conda-forge
occt         7.7.0  conda-forge
ocp          7.7.0.1  conda-forge
```

Using:

adam-urbanczyk, 2023-07-25:
I cannot reproduce on master with OCP 7.7.1

[image]

lorenzncode, 2023-07-26:
I'm getting mixed results.

In local testing, I can reproduce the error on Linux (master with OCP7.7.1, also CQ2.3.1 with OCP7.7.0).
I could not reproduce it on Windows (tested with OCP7.7.0) locally.

In a test on ci.appveyor.com, Windows and Linux pass. MacOS fails with `Standard_Failure`:

https://ci.appveyor.com/project/lorenzncode/cadquery/builds/47638541

bladesaber, 2023-07-26:
After I change the version of ocp to 7.7.1, it works. Maybe it is just a version bug. Thanks very much.

bladesaber, 2023-07-26:
Sometimes, the sweep function seems strange, like the result below and the correct one should be smooth:
[image]

why would cause that?

adam-urbanczyk, 2023-07-26:
Can you share the code?