CadQuery/cadquery #1383: Extrude with taper on any Workplane return a wrong extrude length

trunglebao, 2023-07-31:
Generic use of the extrude(until = x, taper = y) command will result in extrude length less than x.

`a = list(range(0,60))
z = []
for angle in a:
    part = cq.Solid(cq.Workplane('XY').rect(1,1).extrude(1, taper = angle).toOCC())
    bb = part.BoundingBox()
    z.append(bb.zlen)`

Running the code above create a bunch of extrude with height 1, with taper ranging from 0 to 60 degree.
Plotting the taper angle (0 to 60 degree) with resulted extrude height (suppose to be 1):
[image]

Only taper = 0 return the correct height = 1, the rest is all less than 1.
Any thought/alternative solution to get the correct height?
Thanks!

trunglebao, 2023-08-08:
That explained it. Thank you very much!
Most of the CAD software that i use set the extrude length to be zlen regardless of the taper amount. While cq use the length of the tapered edge.

[image]

Regardless, very useful library. Hat off to you guys.

adam-urbanczyk, 2023-08-08:
I think we need to fix this. Me expectation was that this param is height, that is confirmed by the docstring:
```
The distance to extrude, normal to the workplane plane
```

jmwright, 2023-08-19:
@trunglebao This has been fixed in the latest commit, (see linked PR) but the CI system will need a few hours to build and deploy.

PR #1388 (2023-08-19): Fix for incorrect tapered extrude height found in #1383
Closes #1383 

It seems that [LocOpe_DPrism](https://dev.opencascade.org/doc/refman/html/class_loc_ope___d_prism.html) is introducing the issue fround in #1383 where the extrude height requested by the user is not respected. I compensated for this in the code and then added a test so we can detect if the behavior ever changes in OCCT.
closed 2023-08-19 by jmwright