CadQuery/cadquery #1386: cylinder union sphere does not working

arthurbg951, 2023-08-14:
Expectations:
make a cylinder using 2 points then add a sphere to p2

Generating:
a sphere and a cylinder sparse

Code:
import cadquery as cq
from scipy.spatial.distance import euclidean

p1 = cq.Vector(457, 732, 0)
p2 = cq.Vector(457, 733, 0)

diameter = 0.47
lenght = euclidean([p1.x, p1.y], [p2.x, p2.y])

cylinder = (
    cq.Workplane("XY", origin=p1)
    .lineTo(p2.x, p2.y)
    .circle(diameter / 2)
    .extrude(lenght)
)

sphere = (
    cq.Workplane("XY", origin=p2)
    .sphere(diameter / 2)
)

combined_object = cylinder.union(sphere)

test_file = "test.step"
cq.exporters.export(combined_object, test_file, exportType='STEP')

lorenzncode, 2023-08-15:
Do you expect the cylinder and sphere to overlap?  

The shapes do not overlap because the center of the cylinder is  `Vector: (685.4999999999999, 1098.5, 0.49999999999999994)`.  
You can find the center value with `cylinder.val().Center()`.

The center of the sphere is p2.

The workplane origin is p1:
```py
cylinder = (
cq.Workplane("XY", origin=p1)
```

The next statement creates a line starting at p1 `(457, 732, 0)`.
The end of the line is p1+p2.  
The midpoint of the line then becomes the center of the bottom face of the cylinder `(685.5, 1098.5, 0)`. 
```py
.lineTo(p2.x, p2.y)
```

Removing `lineTo`, the center of the cylinder becomes  `(457, 732, 0.5)`.  Still the shapes do not overlap.
To overlap the shapes you could change the origin of the shapes, or move the shapes.

Replacing `.lineTo(p2.x, p2.y)` with a relative move say `.move(0, 1)` results in overlap effectively the same as if setting the origin of both cylinder and shape to p2.

arthurbg951, 2023-08-16:
Tanks for help. 

My problem was with 3d coordinate systems. Depending what Workplane was choosen the behavior changes alot to extrude, like using 2 cylinders in diferent axes like the code below. 

Code:
`import cadquery as cq
from scipy.spatial.distance import euclidean

diameter = 0.47

p1 = cq.Vector(457, 732, 0)
p2 = cq.Vector(457, 733, 0)
p3 = cq.Vector(458, 733, 0)

dist1 = euclidean([p1.x, p1.y], [p2.x, p2.y])
dist2 = euclidean([p2.x, p2.y], [p3.x, p3.y])

print(p1, p2, p3, dist1, dist2, diameter)

sphere1 = (
    cq.Workplane('XY', origin=p1)
    .sphere(diameter / 2)
)

sphere2 = (
    cq.Workplane('XY', origin=p2)
    .sphere(diameter / 2)
)

sphere3 = (
    cq.Workplane('XY', origin=p3)
    .sphere(diameter / 2)
)

cylinder1 = (
    cq.Workplane('XZ', origin=p1)
    .circle(diameter / 2)
    .extrude(-dist1)
)

cylinder2 = (
    cq.Workplane('YZ', origin=p2)
    .circle(diameter / 2)
    .extrude(dist2)
)

union = (
    sphere1
    .union(cylinder1)
    .union(sphere2)
    .union(cylinder2)
    .union(sphere3)
)

combined_file = "combined.step"
cq.exporters.export(union, combined_file, exportType="STEP")
`
That behavior was what i was looking for.

arthurbg951, 2023-08-16:
I will check this as solved. Tanks.

closed 2023-08-16 by arthurbg951