CadQuery/cadquery #1389: Import file from AWS bucket

michaelnicol, 2023-08-19:
I am using cadquery to convert files from STEP to STL.

For this, the files are stored on an AWS server with public-read.

To import, I use the following statement:

```
afile = cq.importers.importStep("https://company_bucket_name.region.amazonaws.com/file.step") # (mock link)
```

The file does exist in the real link, as going to it downloads the file. However, when attempting to import via cadquery, I get:

```
ValueError: STEP File could not be loaded
```

jmwright, 2023-08-19:
The importer does not have a web client built into it. You'll have to use something like the `requests` module in Python or wrap your CadQuery script in a shell script that uses wget, curl, etc to download the file first.

michaelnicol, 2023-08-19:
Could CadQuery in a AWS lambda to convert the file on the AWS cloud?

jmwright, 2023-08-20:
I'm only somewhat familiar with AWS lambdas. I think it's possible, but I can't provide provide any guidance on how to do it. If the lambda will let you import the requests module, you might be able to do this:

https://www.w3schools.com/python/module_requests.asp