CadQuery/cadquery #1392: rotate a cylinder using 2 points as reference

arthurbg951, 2023-08-28:
I wanna create a cylinder between 2 points the code below but when i rotate it the cylinder does not connect all spheres.

My problem is to define  axisStartPoint and axisEndPoint in rotate funciton for this set of points (exemple). How can i make this cylinder connnect p1 and p2 with rotation? 

The code must works even with p1=cq.Vector(0,0,0) and p1=cq.Vector(5,0,5)

```
import cadquery as cq
import os
from scipy.spatial.distance import euclidean
import math

p1 = cq.Vector(2, 0, 2)
p2 = cq.Vector(1, 0, 1)

direction = p2 - p1

diameter = 0.47
height = euclidean([p1.x, p1.z], [p2.x, p2.z])
angle = math.degrees(cq.Vector(direction.x, 0, 0).getAngle(direction))

print(f'direction={direction} diameter={diameter} height={height} angle={angle}')

cylinder = (
    cq.Workplane("XZ", origin=cq.Vector(p1.x, p1.z, p1.y))
    .circle(diameter / 2)
    .extrude(height)
)

# ERROR PART
rotated_cylinder = cylinder.rotate(cq.Vector(p1.x, p1.z, p1.y), cq.Vector(0, 0, 0), angle)

sphere1 = (
    cq.Workplane("XY")
    .moveTo(p1.x, p1.z)
    .sphere(diameter / 2)
)

sphere2 = (
    cq.Workplane("XY")
    .moveTo(p2.x, p2.z)
    .sphere(diameter / 2)
)

obj = rotated_cylinder.union(sphere1).union(sphere2)

output_dir = 'output'
os.mkdir(output_dir) if not os.path.exists(output_dir) else ...

combined_file = os.path.join(output_dir, "diagonal_cylinder.step")
cq.exporters.export(obj, combined_file, exportType="STEP")

```

arthurbg951, 2023-08-28:
To simplify the problem:
I have 2 points and i wanna make a cylinder between them.

The best way to understand the set of points is using this code:

```
import cadquery as cq
import os
from scipy.spatial.distance import euclidean
import math

p1 = cq.Vector(1, 1)
p2 = cq.Vector(0, 0)

direction = p2 - p1

diameter = 0.47
height = euclidean([p1.x, p1.y], [p2.x, p2.y])
angle = math.degrees(cq.Vector(1, 0, 0).getAngle(direction))

print(f'direction={direction} diameter={diameter} height={height} angle={angle}')

cylinder = (
    cq.Workplane("XZ", origin=cq.Vector(p1.x, 0))
    .circle(diameter / 2)
    .extrude(height)
)
```

Depending what point is p1 the direction to rotate cylinder also changes. In this case the cylinder is being extruded in X+ using p1 as basepoint (which means in XY axis the cylinder can be represented as vector(0,height)). This means the angle between vector (1,0) until direction is 135 degrees (using p1 and p2 as this code exemple above). Changing the p1=(0,0) and p2=(1,1) the angle between vector(1,0) and direction is 45 degrees.

Based in what i have write before i does not understand what is: axisStartPoint, axisEndPoint and angleDegrees

Shold angleDegrees be 135 or 45 (depending what is p1 and p2)? Should axisStartPoint and axisEndPoint be p1 and p2?

lorenzncode, 2023-08-28:
Try the following in the first example:
```py
rotated_cylinder = cylinder.rotate((p1.x, p1.z, p1.y), (p1.x, p1.z, p1.y+1),  -angle)
```

To visualize the axis of rotation:
```py
axis = cq.Workplane("XZ", origin=(p1.x, p1.z, p1.y)).lineTo(0, 1)
# or 
axis = cq.Edge.makeLine((p1.x, p1.z, p1.y), (p1.x, p1.z, p1.y+1))
```

arthurbg951, 2023-08-31:
Solved with, tank you:

```
import cadquery as cq
import os
from scipy.spatial.distance import euclidean
import math

p1 = cq.Vector(2, 1)
p2 = cq.Vector(1, 1)

direction = p2 - p1

diameter = 0.47
height = euclidean([p1.x, p1.y], [p2.x, p2.y])
angle = math.degrees(cq.Vector(0, 1).getAngle(direction))

print(f'direction={direction} diameter={diameter} height={height} angle={angle}')

# Extrude function aways extrude leaving the plane (for plane XZ extruding are in Y-).
# This justify -height. To invert extrude way to Y+.
cylinder = (
    cq.Workplane("XZ", origin=cq.Vector(p1.x, p1.y, p1.z))
    .circle(diameter / 2)
    .extrude(-height)
)

# angleDegrees must be changed to the desired rotation.
# The default rotation for rotate function is counter clockwise.
direction_of_rotation = -1 * math.copysign(1, direction.x)
rotated_cylinder = cylinder.rotate((p1.x, p1.y, p1.z), (p1.x, p1.y, p1.z + 1), direction_of_rotation * angle)

sphere1 = (
    cq.Workplane("XY")
    .moveTo(p1.x, p1.y)
    .sphere(diameter / 2)
)

sphere2 = (
    cq.Workplane("XY")
    .moveTo(p2.x, p2.y)
    .sphere(diameter / 2)
)

obj = rotated_cylinder.union(sphere1).union(sphere2)

output_dir = 'output'
os.mkdir(output_dir) if not os.path.exists(output_dir) else ...

combined_file = os.path.join(output_dir, "diagonal_cylinder.step")
cq.exporters.export(obj, combined_file, exportType="STEP")

```

closed 2023-08-31 by arthurbg951