CadQuery/cadquery #1395: Support each with `cq.Shape` and `cq.Workplane` in addition to callbacks

adam-urbanczyk, 2023-08-31:
E.g.
```python
b = cq.Workplane().box(2,2,2)
s = cq.Workplane().sphere(0.5)

res = b.faces().each(s)
```

dov, 2024-04-28:
@adam-urbanczyk  In continuation of your comment on #1574 , I would add `cq.Wire` in addition to `cq.Shape` and `cq.Workplane` to this issue.

adam-urbanczyk, 2024-04-28:
`cq.Wire` **_is a_** `cq.Shape`

dov, 2024-04-28:
@adam-urbanczyk I got it working for a `Shape` by renaming the argument `arg` and dispatching on its type. However I need some guidance about what semantics you expect for a Workspace. Just moving the workspace by the following lambda obviously does not work:

```python
        if isinstance(arg, Workplane):
            wp = arg
            callback = lambda v : wp.move(v)
```

adam-urbanczyk, 2024-04-28:
Something like this:

```python
v.moved(loc) for v in arg.vals() if isinstance(v, Shape)
```

dov, 2024-04-29:
> Something like this:
> 
> ```python
> v.moved(loc) for v in arg.vals() if isinstance(v, Shape)
> ```

Thanks @adam-urbanczyk . In my PR I used the following. Let me know this is what you intended.

```python
                res = [
                    v.moved(p).move(loc)
                    for v in arg.vals()
                    for p in pnts
                    if isinstance(v, Shape)
                ]
```

lorenzncode, 2024-08-16:
I think this can be closed.  One follow up issue though is `cutEach` typing.  

This now works but fails mypy:
```py
b = cq.Workplane().box(2, 2, 2)
s = cq.Workplane().sphere(0.5)

res = b.faces().cutEach(s)
```

```
test2.py:7: error: Argument 1 to "cutEach" of "Workplane" has incompatible type "Workplane"; expected "Callable[[Location], Shape]"  [arg-type]
```

adam-urbanczyk, 2024-08-19:
We can probably deprecate `cutEach` it is the same as `eachpoint(..., mode='cut')`.

closed 2024-08-19 by adam-urbanczyk

lorenzncode, 2024-08-19:
Yes, agree deprecating `cutEach` makes sense.