CadQuery/cadquery #1399: Inverted normals when using shell and fillet

blastrock, 2023-09-03:
The following code with a negative shell works fine:

```
test = (
    cq.Workplane("XY")
    .box(10, 10, 10)
    .faces(">Z")
    .shell(-1, "intersection")
    .edges("|Z")
    .fillet(1)
)
show_object(test)
```

[image]

However, if we make the shell positive, the normals get inverted and we get "holes" in the shape

```
test = (
    cq.Workplane("XY")
    .box(10, 10, 10)
    .faces(">Z")
    .shell(1, "intersection") # <----- positive shell
    .edges("|Z")
    .fillet(1)
)
show_object(test)
```

[image]

It works fine without the fillet.

Version: 2.3.1_43_gad46448

jmwright, 2023-09-06:
I am unable to reproduce the visual error with the VTK-based show feature, but I can with CQ-editor (latest master). Also, when I export to STL with negative shell there are no mesh errors, but when I export to STL with positive shell, there are over 1500 errors that are auto-repaired.

VTK-based show:
[image]

CQ-editor:
[image]

STL Errors:
[image]

adam-urbanczyk, 2023-09-29:
`test.val().isValid()` returns `False`. In general the shell function is not always reliable.  You can using `test.val().fix()`