CadQuery/cadquery #1405: TestDxfDocument.test_spline fails with unsupported operand

DoubleDipl, 2023-09-21:
Dear all,

I try to compile cadquery on arm64. I already managed to get to compiling cadquery but one of the tests fails:

```
 ============================= test session starts ==============================
platform linux -- Python 3.10.12, pytest-7.4.2, pluggy-1.3.0 -- $PREFIX/bin/python3.10

[...]

=================================== FAILURES ===================================
_________________________ TestDxfDocument.test_spline __________________________

self = <tests.test_exporters.TestDxfDocument testMethod=test_spline>

     def test_spline(self):
         pts = [(0, 0), (0, 0.5), (1, 1)]
         workplane = (
             Workplane().spline(pts).close().extrude(1).edges("|Z").fillet(0.1).section()
         )

         plane = workplane.plane
         shape = toCompound(workplane).transformShape(plane.fG)
         edges = shape.Edges()

         result_type, result_attributes = DxfDocument._dxf_spline(edges[0], plane)

         expected_type, expected_attributes = (
             "SPLINE",
             {
                 "control_points": [
                     (-0.032010295564216654, 0.2020130195642037, 0.0),
                     (-0.078234124721739, 0.8475143728081896, 0.0),
                     (0.7171193004814275, 0.9728923786984539, 0.0),
                 ],
                 "order": 3,
                 "knots": [
                     0.18222956891558767,
                     0.18222956891558767,
                     0.18222956891558767,
                     1.416096480384525,
                     1.416096480384525,
                     1.416096480384525,
                 ],
                 "weights": None,
             },
         )

         self.assertEqual(expected_type, result_type)
 >       self.assertAlmostEqual(
             expected_attributes["control_points"], result_attributes["control_points"]
         )
E       TypeError: unsupported operand type(s) for -: 'list' and 'list'

 tests/test_exporters.py:418: TypeError
```

I assume that this is not an arm-specific issue. I found some topics on the web, that almost equal doesn't support lists of floats. I wonder why this test seems to be passing in the official build pipeline. Any idea what could be the reason in my case?

Best regards

adam-urbanczyk, 2023-09-29:
Are you sure that all the deps have correct versions. Recent CI is green, so most likely the issue is on your side.

DoubleDipl, 2023-09-30:
Sorry, I'm really not an expert and I cannot exclude it is some sort of dependency issue. But as far as I understand `TestDxfDocument` is derived from `BaseTest` which itself is derived from `unittest.TestCase`.

As far as I can see `unittest.TestCase` does define `assertAlmostEqual` according to [Python Docs](https://docs.python.org/3/library/unittest.html#unittest.TestCase.assertAlmostEqual), which is performing a subtraction of the given arguments and rounding afterwards. I don't see `BaseTest` overriding this method, hence, I assume this is what is happening in this test.

Am I wrong with assuming python does not support subtraction of lists? Because `expected_attributes["control_points"]` is a list of tuples, for me this would clearly be the root cause. However, I wonder why CI is passing.

I found `pytest.approx` to be able of handling lists. However, I really would be interested in finding the real root cause for this issue.

Maybe someone can provide a full list of packages installed in build/test environment for me to compare with the packages installed on my side?

lorenzncode, 2023-09-30:
I think it passes in CI because of the shortcut:
https://github.com/python/cpython/blob/446c81c3c3a5e82c1df475b8a98009aeb5b5433e/Lib/unittest/case.py#L898-L900

I'd propose to update the test to something like this:
```py
        for expected, result in zip(expected_attributes["control_points"], result_attributes["control_points"]):
            assert expected == approx(result)
```

Are the actual test results on arm "appproximately" equal here or is it a real failure?

DoubleDipl, 2023-10-01:
Tank you for your investigation. I ran the build process with suggested modification. It seems to be working, but now another failure arises. I think this might be somehow related as basically it seems to be a floating point precision issue (as also for not triggering the mentioned shortcut):

```
#31 198.9 =================================== FAILURES ===================================
#31 198.9 _________________________ TestDxfDocument.test_spline __________________________
#31 198.9 
#31 198.9 self = <tests.test_exporters.TestDxfDocument testMethod=test_spline>
#31 198.9 
#31 198.9     def test_spline(self):
#31 198.9         pts = [(0, 0), (0, 0.5), (1, 1)]
#31 198.9         workplane = (
#31 198.9             Workplane().spline(pts).close().extrude(1).edges("|Z").fillet(0.1).section()
#31 198.9         )
#31 198.9     
#31 198.9         plane = workplane.plane
#31 198.9         shape = toCompound(workplane).transformShape(plane.fG)
#31 198.9         edges = shape.Edges()
#31 198.9     
#31 198.9         result_type, result_attributes = DxfDocument._dxf_spline(edges[0], plane)
#31 198.9     
#31 198.9         expected_type, expected_attributes = (
#31 198.9             "SPLINE",
#31 198.9             {
#31 198.9                 "control_points": [
#31 198.9                     (-0.032010295564216654, 0.2020130195642037, 0.0),
#31 198.9                     (-0.078234124721739, 0.8475143728081896, 0.0),
#31 198.9                     (0.7171193004814275, 0.9728923786984539, 0.0),
#31 198.9                 ],
#31 198.9                 "order": 3,
#31 198.9                 "knots": [
#31 198.9                     0.18222956891558767,
#31 198.9                     0.18222956891558767,
#31 198.9                     0.18222956891558767,
#31 198.9                     1.416096480384525,
#31 198.9                     1.416096480384525,
#31 198.9                     1.416096480384525,
#31 198.9                 ],
#31 198.9                 "weights": None,
#31 198.9             },
#31 198.9         )
#31 198.9     
#31 198.9         self.assertEqual(expected_type, result_type)
#31 198.9         for expected, result in zip(expected_attributes["control_points"], result_attributes["control_points"]):
#31 198.9             assert expected == approx(result)
#31 198.9         self.assertEqual(expected_attributes["order"], result_attributes["order"])
#31 198.9 >       self.assertEqual(expected_attributes["knots"], result_attributes["knots"])
#31 198.9 E       AssertionError: Lists differ: [0.18[55 chars]67, 1.416096480384525, 1.416096480384525, 1.416096480384525] != [0.18[55 chars]67, 1.4160964803845248, 1.4160964803845248, 1.4160964803845248]
#31 198.9 E       
#31 198.9 E       First differing element 3:
#31 198.9 E       1.416096480384525
#31 198.9 E       1.4160964803845248
#31 198.9 E       
#31 198.9 E         [0.18222956891558767,
#31 198.9 E          0.18222956891558767,
#31 198.9 E          0.18222956891558767,
#31 198.9 E       -  1.416096480384525,
#31 198.9 E       ?                  ^
#31 198.9 E       
#31 198.9 E       +  1.4160964803845248,
#31 198.9 E       ?                  ^^
#31 198.9 E       
#31 198.9 E       -  1.416096480384525,
#31 198.9 E       ?                  ^
#31 198.9 E       
#31 198.9 E       +  1.4160964803845248,
#31 198.9 E       ?                  ^^
#31 198.9 E       
#31 198.9 E       -  1.416096480384525]
#31 198.9 E       ?                  ^
#31 198.9 E       
#31 198.9 E       +  1.4160964803845248]
#31 198.9 E       ?                  ^^
#31 198.9 
#31 198.9 tests/test_exporters.py:421: AssertionError
```

PR #1414 (2023-11-26): test: Fix comparison of tuple of floats in TestDxfDocument::test_spline
Use pytest.approx instead of unnittest.assertEqual assertAlmostEqual.

Resolves #1405
closed 2023-11-26 by adam-urbanczyk