CadQuery/cadquery #1410: Generating a minimum step file with a lot of geometries

arthurbg951, 2023-09-30:
I'm doing a processing with a lot of points and I need to create a set of cylinders with them. For now, I'm using a stepfile to create the full geometry but the file created is to slow. Any sugestion to improve the geometry creation? The step file created is huge (50-60mb)

```
# Iterating in a set of points (alot of them)
for idx, fib in enumerate(fibers):
        seg = np.array(fib)
        y_coords, x_coords = seg[:, 0] * -scale_factor, seg[:, 1] * scale_factor

        diameter = 0.47

        obj = cq.Workplane("XY")

        for idx in range(len(x_coords) - 1):
            p1 = cq.Vector(x_coords[idx], y_coords[idx])
            p2 = cq.Vector(x_coords[idx + 1], y_coords[idx + 1])

            direction = p2 - p1

            height = euclidean([p1.x, p1.y], [p2.x, p2.y])
            angle = math.degrees(cq.Vector(0, 1).getAngle(direction))

            direction_of_rotation = -1 * math.copysign(1, direction.x)
            cylinder = (
                cq.Workplane("XZ", origin=cq.Vector(p1.x, p1.y, p1.z))
                .circle(diameter / 2)
                .extrude(-height)
                .rotate((p1.x, p1.y, p1.z), (p1.x, p1.y, p1.z + 1), direction_of_rotation * angle)
            )

            obj.add(cylinder)

        output_obj.add(obj)

combined_file = os.path.join('output', "multi_fibers.step")
cq.exporters.export(output_obj, combined_file, exportType="STEP")
```

lorenzncode, 2023-09-30:
Try creating the cylinder with:
```py
cylinder = cq.Solid.makeCylinder(diameter / 2, height, p1, direction)
```

There is an option that can reduce STEP file size:
```py
cq.exporters.export(output_obj, combined_file, exportType="STEP", opt={"write_pcurves": 0})
```

adam-urbanczyk, 2023-10-01:
Are some cylinders identical? If so, using `located` instead of creating them always from scratch should result in a smaller step too.

arthurbg951, 2023-10-01:
@adam-urbanczyk can you give me an exemple?

arthurbg951, 2023-10-01:
> Try creating the cylinder with:
> 
> ```python
> cylinder = cq.Solid.makeCylinder(diameter / 2, height, p1, direction)
> ```
> 
> There is an option that can reduce STEP file size:
> 
> ```python
> cq.exporters.export(output_obj, combined_file, exportType="STEP", opt={"write_pcurves": 0})
> ```

The opt setting makes a good difference. I will try make the cylinder using cq.Solid and compare results.

adam-urbanczyk, 2023-10-01:
> @adam-urbanczyk Not always, length can be 1 or 1.4142. Direction also could change, but it will always be horizontal, vertical or diagonal. The diameter never change. Can you give me an example using **located** ?

If *some* are identical (i.e. having same radius and height):

```python
#create cylinders
s1 = cq.Workplane().circle(r).extrude(h)
...
# loop over locations and identical cylinders
...
res.add(s1.val().moved(cq.Location((x,y,z), (a,b,c)))
...
```

arthurbg951, 2023-10-13:
None of our help could solve my problem. Because it have alot of points..

This is an image what I'm working and the set of points come from here. What should I do to make cylinders with this coordinates? (My current solution is to create an cylinder and 2 spheres at start  and end point foreach coordinate. thats what causing the huge stepfile)

[image]

arthurbg951, 2023-10-13:
Here is one exemple after processing.. 

[image]

But I need a lot of them.

arthurbg951, 2023-10-18:
Does anyone have any idea?

adam-urbanczyk, 2023-10-20:
So did instancing (i.e. using moved on identical shapes with different locations) result in no change of the file size? If it was done correctly and there is no significant gain (e.g. there are not so many geometries the repeat, though from your picture it seems to not be the case), I have no ideas regarding STEP. Depending on your application you might consider exporting to a binary brep file. Another question would be, why do you need to serialize, if you need to e..g mesh with netgen or gmsh it is possible to share the geometry in-memory.