CadQuery/cadquery #1412: show_object sometimes does not work

adam-urbanczyk, 2023-10-03:
If it helps, I get the same on Fedora 38 kind of error with CadQuery installed in a Python 3.11 virtual environment. Using the following code:
```Python
import cadquery as cq
from cadquery.vis import show

#d = 10  # box is visible as expected
d = 100  # empty window displayed; after 'r' key or zoom with middle mouse the box is displayed

result = cq.Workplane().box(d, d, d)

show(result)
```
With the #1394 branch I also got:
```
2023-10-02 23:51:41.837 (   0.491s) [        4D584740]vtkDemandDrivenPipeline:661    ERR| vtkCompositeDataPipeline (0x56454faba5e0): Input port 0 of algorithm vtkTriangleFilter (0x56454ff2f010) has 0 connections but is not optional.
Traceback (most recent call last):
  File "/home/Projects/cadquery/cadquery_vis.py", line 9, in <module>
    show(result)
  File "/home/Projects/cadquery/lib64/python3.11/site-packages/cadquery/vis.py", line 82, in show
    win.SetSize(*win.GetScreenSize())
TypeError: vtkRenderWindow.SetSize() argument after * must be an iterable, not NoneType
```
Adding a `print` statement for `win.GetScreenSize()` and it returns `None`. The workaround from https://github.com/CadQuery/cadquery/pull/1366#issuecomment-1695614005 results in:
```
2023-10-03 00:08:51.661 (   0.487s) [        55B74740]vtkDemandDrivenPipeline:661    ERR| vtkCompositeDataPipeline (0x5618c5e38540): Input port 0 of algorithm vtkTriangleFilter (0x5618c6028e70) has 0 connections but is not optional.
X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  3 (X_GetWindowAttributes)
  Resource id in failed request:  0x0
  Serial number of failed request:  8
  Current serial number in output stream:  9
```
I'm running on X11, maybe Wayland will work?

_Originally posted by @Jopie01 in https://github.com/CadQuery/cadquery/issues/1366#issuecomment-1743842557_

Jopie01, 2023-10-03:
Ok, after a bit more digging it seems something goes wrong with OCP. The line https://github.com/CadQuery/cadquery/blob/d7801273704134fdb172e3c6ac22d34d326e2800/cadquery/occ_impl/shapes.py#L1335 returns `None` so no data is returned.

I downgraded `cadquery-ocp` to version `7.7.0` with ` pip install --upgrade cadquery-ocp==7.7.0` and voilà ..... I got a cube! 

[image]

rohit-kumar-j, 2023-10-04:
Nice! This works for me on windows.

adam-urbanczyk, 2023-10-05:
OK, so all this with the pip package. Can you also reproduce this with the OCP conda package from conda-forge?