CadQuery/cadquery #1421: hole() in wrong direction

justbuchanan, 2023-10-19:
cadquery version: cq-editor [nightly release](https://github.com/CadQuery/CQ-editor/releases/tag/nightly) from May 14

code:

```.py
# make a cylinder by revolving a rectangle around the Z axis
obj = cq.Workplane("XZ").lineTo(10, 0).lineTo(10, 10).lineTo(0, 10).close().revolve()
# put a hole in the top
obj = obj.faces(">Z").hole(3)
```

result:
[image]

expected result (I made this using `cut()` and a cylinder):

[image]

Jojain, 2023-10-19:
This is the correct behavior you want to add a `workplane()` call after your face selection

justbuchanan, 2023-10-19:
That worked, thank you!