CadQuery/cadquery #1427: Support outputting a STEP file with per-face colors

mkovaxx, 2023-10-27:
I need to set per-face colors on programmatically created shapes, then export into STEP format. Please help me find a way to do this. I'm somewhat familiar with the underlying C++ OCCT, and I'm more than happy to contribute code. I'm filing this issue so we can discuss and align on the best approach here.

Here's a self-contained demo of the approach I found so far:
https://github.com/mkovaxx/cadquery_fused_assembly/tree/4572a63616a614f3920c859b5224c8d4aaed3d64

I create each face as a separate shape, then pass it to the `Assembly.add` method along with its desired color. Saving the assembly as STEP then contains the colors as expected.

For details, see this section:
https://github.com/mkovaxx/cadquery_fused_assembly/blob/4572a63616a614f3920c859b5224c8d4aaed3d64/src/tubes.py#L22-L37

However, this approach has two severe limitations.

First, the output contains topologically disconnected shells. I expected this to be resolved by setting `ExportModes.FUSED` on the `Assembly.save` method, unfortunately that does not help.

Second, the output contains no volumes. There doesn't seem to be a way to get around this without extending CadQuery in some way.

adam-urbanczyk, 2023-10-27:
So on OCCT level you have to use sub-shapes (see fused assy export). This could be added to the CQ assy class via an optional field mapping sub-shapes to color:
```python
colors: Dict[Shape, Color]
```
Additionally, assy export would have to be updated to actually make use of this property.

mkovaxx, 2023-11-09:
@adam-urbanczyk Thanks for proposing an approach!

I'll write down what I understood just to make sure that it is correct.

1. Extend `Assembly` with a `subshape_colors: Dict[Shape, Color]` field.
2. Extend `exportAssembly` to encode `subshape_colors` into the `TDocStd_Document` before it is written as a STEP file.

Is that correct?

Based that, I think these are the places that require modifications:

1. Add `subshape_colors: Dict[Shape, Color]` in here:
https://github.com/CadQuery/cadquery/blob/7bc8c00dc1a81633daea4810e04762d61a2abd5a/cadquery/assembly.py#L82-L90

2. Propagate `subshape_colors` into the call to `exportAssembly` here:
https://github.com/CadQuery/cadquery/blob/7bc8c00dc1a81633daea4810e04762d61a2abd5a/cadquery/assembly.py#L486-L487

3. Encode `subshape_colors` into `doc` here, between line 86 and 88:
https://github.com/CadQuery/cadquery/blob/7bc8c00dc1a81633daea4810e04762d61a2abd5a/cadquery/occ_impl/exporters/assembly.py#L82-L95

Did I miss anything?

adam-urbanczyk, 2023-11-09:
So (1) sounds fine. There should be (1.5) which is adding some kind of method (methods?) to set colors e.g. based on selectors. (2) is not required - you don't need to propagate anything, it is already a field. (3) actually requires modifying `toCAF()` and is likely going to be more complicated than you wrote.

closed 2025-10-04 by adam-urbanczyk