CadQuery/cadquery #1429: Intermediate path segments not being rendered until `close` call

vrzh, 2023-10-30:
This is probably more of a question, but also maybe a bug.

I really like the "automatic reload and preview" inside CQ-Editor.

I am trying to build up a complicated path using many `lineTo` calls, and want to have a preview as I am in the middle of this. However, in the CQ-Editor preview window, I don't see the entire path (only the last segment).

If I want to see the path so far, I need to call `close` at the end, and then it renders. But, I am not finished with the path yet and so don't want to close it.

This seems not related to CQ-Editor itself, because if you generate an SVG output for attached example, you get the same effect.

I am using `CadQuery` and `CQ-Editor` via `pip`.

```py
import cadquery as cq

# without the "close", only the last segment "lineTo(2,1)" is shown
result = cq.Workplane("XY").lineTo(1,2).lineTo(3,3).lineTo(2,1)

# with the "close", all segments are shown
# result = cq.Workplane("XY").lineTo(1,2).lineTo(3,3).lineTo(2,1).close()

with open('t.svg', 'w') as f:
    f.write(result.toSvg())

show_object(result)

```

Screenshot:

[image]

SVG output:
[image]

adam-urbanczyk, 2023-10-31:
So this is not a bug. For `Workplane` only top of the stack is displayed.

vrzh, 2023-10-31:
> So this is not a bug. For `Workplane` only top of the stack is displayed.

Ahh, yes. That makes sense. Thank you.

vrzh, 2023-10-31:
> Call `consolidateWires()` to render the intermediate result.
> 
> ```python
> wi0 = result.consolidateWires()
> show_object(wi0)
> ```
> 
> If you are working with Sketch, you can view the intermediate result like this.:
> 
> ```python
> import cadquery as cq
> 
> s0 = (
>     cq.Sketch()
>     .segment((0, 0), (1, 2))
>     .segment((3, 3))
>     .segment((2, 1))
>     #.close()
>     #.assemble()
> )
> 
> show_object(s0)
> ```

Perfect, I will use this. Thank you.