CadQuery/cadquery #1431: 'sweep' end in 'TypeError: typing_extensions.Literal cannot be used with isinstance()'

mryndzionek, 2023-11-01:
Simple example:
```py
import cadquery as cq

r = 0.5  # Radius of the helix
p = 0.4  # Pitch of the helix - vertical distance between loops
h = 2.4  # Height of the helix - total height

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
helix = cq.Workplane(obj=wire)

# Final result: A 2D shape swept along a helix.
result = (
    cq.Workplane("XZ")  # helix is moving up the Z axis
    .center(r, 0)  # offset rect
    .rect(0.15, 0.15)
    .sweep(helix, isFrenet=True)  # Frenet keeps orientation as expected
)

show_object(result)
```

end with:

```
Traceback (most recent call last):
  File "/cq-editor/lib/python3.9/site-packages/cq_cli/main.py", line 36, in build_and_parse
    raise (build_result.exception)
  File "/cq-editor/lib/python3.9/site-packages/cadquery/cqgi.py", line 119, in build
    exec(c, env)
  File "<cqscript>", line 32, in <module>
  File "/cq-editor/lib/python3.9/site-packages/cadquery/cq.py", line 3196, in sweep
    r = self._sweep(
  File "/cq-editor/lib/python3.9/site-packages/cadquery/cq.py", line 3804, in _sweep
    thisObj = Solid.sweep(f, p, makeSolid, isFrenet, mode, transition)
  File "/cq-editor/lib/python3.9/site-packages/cadquery/utils.py", line 50, in __call__
    return super().__call__(*args, **kwargs)
  File "/cq-editor/lib/python3.9/site-packages/multimethod/__init__.py", line 326, in __call__
    func = self[tuple(func(arg) for func, arg in zip(self.type_checkers, args))]
  File "/cq-editor/lib/python3.9/site-packages/multimethod/__init__.py", line 326, in <genexpr>
    func = self[tuple(func(arg) for func, arg in zip(self.type_checkers, args))]
  File "/cq-editor/lib/python3.9/site-packages/multimethod/__init__.py", line 131, in get_type
    if not isinstance(arg, self.__origin__):  # no need to check subscripts
  File "/usr/lib/python3.9/typing.py", line 338, in __instancecheck__
    raise TypeError(f"{self} cannot be used with isinstance()")
TypeError: typing_extensions.Literal cannot be used with isinstance()
```

mryndzionek, 2023-11-02:
> I am not able to reproduce this error. What version of CadQuery are you running, and how did you install it?

2.3.1, installed using pip. Tested on Python 3.8 and 3.9.

jmwright, 2023-11-02:
Ok, I can reproduce with CadQuery 2.3.1, but cannot with the latest git version, so it has been fixed. I don't have time to track down exactly where right now, but I'm guessing it was during the rework of topology traversal.

Can you install CadQuery from git and try again?

mryndzionek, 2023-11-02:
> Ok, I can reproduce with CadQuery 2.3.1, but cannot with the latest git version, so it has been fixed. I don't have time to track down exactly where right now, but I'm guessing it was during the rework of topology traversal.
> 
> Can you install CadQuery from git and try again?

I installed using:
```
pip install git+https://github.com/CadQuery/cadquery.git
```
Reported version is: 2.4.0.dev0, but I still hit the same error.

adam-urbanczyk, 2023-11-02:
Could you maybe specify your env completely?

jmwright, 2023-11-02:
Your code above has `show_object`. Are you running it in CQ-editor? If so, how did you install it? It could be running the version of CQ that was bundled with CQ-editor.

mryndzionek, 2023-11-02:
> Could you maybe specify your env completely?

What exactly do you need? I've installed CQ-Editor in a fresh venv. Added:
```
    install_requires=[
        "logbook>=1",
        "path>=16",
        "PyQt5>=5",
        "pyqtgraph==0.12",
        "requests>=2,<3",
        "spyder>=5,<6")
```
to `setup.py`.

mryndzionek, 2023-11-02:
> Your code above has `show_object`. Are you running it in CQ-editor? If so, how did you install it? It could be running the version of CQ that was bundled with CQ-editor.

Yes, I'm running in CQ-editor. Fresh install in clean venv. I'm printing version from my script, so pretty sure CQ-editor uses ` 2.4.0.dev0`. I also tried `cq-cli` - same error.

adam-urbanczyk, 2023-11-03:
> > Could you maybe specify your env completely?
> 
> What exactly do you need? I've installed CQ-Editor in a fresh venv. Added:

This `conda list` or this `pip list`.

mryndzionek, 2023-11-03:
> > > Could you maybe specify your env completely?
> > 
> > 
> > What exactly do you need? I've installed CQ-Editor in a fresh venv. Added:
> 
> This `conda list` or this `pip list`.

[pip_list.txt](https://github.com/CadQuery/cadquery/files/13247301/pip_list.txt)

adam-urbanczyk, 2023-11-03:
Does downgrading multimethod to 1.9.1 solve the issue?

mryndzionek, 2023-11-03:
> Does downgrading multimethod to 1.9.1 solve the issue?

Okay, yes, it does. Thanks!

adam-urbanczyk, 2023-11-03:
Yes, I think so

adam-urbanczyk, 2023-11-07:
@mryndzionek I cannot reproduce your issue locally.

mryndzionek, 2023-11-07:
> @mryndzionek I cannot reproduce your issue locally.

What can I say, I posted the 'pip list' above. With this environment I definitely hit the error.

adam-urbanczyk, 2023-11-07:
I do get a regression on the tests though, so there is some kind of problem.

lorenzncode, 2023-11-08:
I can reproduce the error with a Python 3.9.18 env: 

<details>
<pre>
(mm110py39) [lorenzn@fedora tmp]$ python test.py 
<cadquery.occ_impl.shapes.Wire object at 0x7f518f81efd0>
Traceback (most recent call last):
  File "/tmp/test.py", line 15, in <module>
    cq.Workplane("XZ")  # helix is moving up the Z axis
  File "/home/lorenzn/tools/mambaforge_230901/envs/mm110py39/lib/python3.9/site-packages/cadquery/cq.py", line 3196, in sweep
    r = self._sweep(
  File "/home/lorenzn/tools/mambaforge_230901/envs/mm110py39/lib/python3.9/site-packages/cadquery/cq.py", line 3804, in _sweep
    thisObj = Solid.sweep(f, p, makeSolid, isFrenet, mode, transition)
  File "/home/lorenzn/tools/mambaforge_230901/envs/mm110py39/lib/python3.9/site-packages/cadquery/utils.py", line 50, in __call__
    return super().__call__(*args, **kwargs)
  File "/home/lorenzn/tools/mambaforge_230901/envs/mm110py39/lib/python3.9/site-packages/multimethod/__init__.py", line 326, in __call__
    func = self[tuple(func(arg) for func, arg in zip(self.type_checkers, args))]
  File "/home/lorenzn/tools/mambaforge_230901/envs/mm110py39/lib/python3.9/site-packages/multimethod/__init__.py", line 326, in <genexpr>
    func = self[tuple(func(arg) for func, arg in zip(self.type_checkers, args))]
  File "/home/lorenzn/tools/mambaforge_230901/envs/mm110py39/lib/python3.9/site-packages/multimethod/__init__.py", line 131, in get_type
    if not isinstance(arg, self.__origin__):  # no need to check subscripts
  File "/home/lorenzn/tools/mambaforge_230901/envs/mm110py39/lib/python3.9/typing.py", line 347, in __instancecheck__
    raise TypeError(f"{self} cannot be used with isinstance()")
TypeError: typing_extensions.Literal cannot be used with isinstance()
(mm110py39) [lorenzn@fedora tmp]$ conda list 
# packages in environment at /home/lorenzn/tools/mambaforge_230901/envs/mm110py39:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohttp                   3.8.6            py39hd1e30aa_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.21.0               hd590300_0    conda-forge
ca-certificates           2023.7.22            hbcca054_0    conda-forge
cadquery                  master          2.3.1_76_g7bc8c00    cadquery
cairo                     1.16.0            ha61ee94_1014    conda-forge
casadi                    3.6.3            py39h4b51453_1    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
curl                      8.1.2                h409715c_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
executing                 2.0.1              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
ezdxf                     1.1.0            py39hda80f44_1    conda-forge
ffmpeg                    5.1.2           gpl_h8dda1f0_106    conda-forge
fftw                      3.3.10          nompi_hc118613_108    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.44.0           py39hd1e30aa_0    conda-forge
freeimage                 3.18.0              hca5c823_12    conda-forge
[... 183 lines cut ...]
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.8.4                h0b41bf4_0    conda-forge
xorg-libxau               1.0.11               hd590300_0    conda-forge
[... cut ...]
xz                        5.2.6                h166bdaf_0    conda-forge
yarl                      1.9.2            py39hd1e30aa_1    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge

</pre>
</details>

I could not reproduce the error with Python 3.10.13, or 3.11.6 with multimethod 1.10.

I confirm I can also reproduce the failed `tests/test_cadquery.py::TestCadQuery::testSweep` with the multimethod 1.10 env (both the 3.10.13 and 3.11.6 envs).