CadQuery/cadquery #1434: Installing CadQuery with Pip3 on arm64 Apple M1

jeremiahrose, 2023-11-03:
Just posting this here for future reference, as I couldn't find this solution anywhere in the README or in past issues on this topic.

I was able to install cadquery using the following steps:

1. Make sure Xcode command line tools and Python3 are installed (assuming you are not using *conda)
2. Upgrade pip like `python3 -m pip install --upgrade pip`
3. Add the following `requirements.txt` to your project:
```
cadquery
cadquery-ocp == 7.7.1
pyparsing >= 2.1.9
sphinx == 5.0.1
sphinx_rtd_theme
black == 19.10b0
click == 8.0.4
mypy
codecov
pytest
pytest-cov
ezdxf
typing_extensions
nptyping == 2.0.1
nlopt
path
casadi
multimethod  >= 1.7,<2.0
typish
```
4. Install the intel versions of these packages: `arch -arch x86_64 pip3 install -r requirements.txt`
5. Run the intel version of cadquery and check that it's working:
```
arch -arch x86_64 python3
>>> import cadquery
>>> cadquery.Workplane('XY').box(1,2,3).toSvg()
```

akoen, 2023-11-03:
I think the "better" way to do this is to use https://github.com/CadQuery/ocp-build-system to extract a python wheel from conda, which involves installing M1 OCP from conda and then extracting the wheel using `python setup.py bdist_wheel`. This method should have much better performance

biggestT, 2023-11-08:
Thank you @akoen! Followed your suggestion and it worked after a while. Had to build wheel for [nlopt](https://pypi.org/project/nlopt/#files) as well since that didn't have pypi dist for arm64 either.

I published the wheels built I bult and some instructions on how to use them with poetry here: https://github.com/generativesolutions/cadquery-dist-macos-arm64

at least the `cadquery.Workplane('XY').box(1,2,3).toSvg()` works now

akoen, 2023-11-08:
Awesome, great work!