CadQuery/cadquery #1438: Loop text around Cylinder

jniclas, 2023-11-07:
Hey,

how can I use a long text line, and loop it multiple times around a cylinder to emboss the text into it. 
The text line would have a small angle, so that it would not overlap if the text is longer than the circumference, but would endup in multiple line on top of each. Like:

[image]

Best regards
Niclas

jmwright, 2023-11-15:
@jniclas This issue might be of interest to you. https://github.com/CadQuery/cadquery/issues/109

jniclas, 2023-11-22:
I don't get it running. I followed https://github.com/CadQuery/cadquery/issues/108, but I don't get the text following the path and being upright...   I stuck at this stage:
[image]
But the extrusion direction should be in the direction of normal of the path part where the letter is positioned... and not all to Y:
[image]
I think I have to change transform/rotate the Vector in here:
```python
 [cq.Solid.extrudeLinear(f[0], cq.Vector(0, 1, 0)) for f in faces_on_path]
```
 but don't now how:
```python
def _faceOnWire(self, path: cq.Wire) -> cq.Face:
    """Reposition a face from alignment to the x-axis to the provided path"""
    path_length = path.Length()

    bbox = self.BoundingBox()
    face_bottom_center = cq.Vector((bbox.xmin + bbox.xmax) / 2, 0, 0)
    relative_position_on_wire = face_bottom_center.x / path_length
    wire_tangent = path.tangentAt(relative_position_on_wire)
    wire_angle = degrees(atan2(wire_tangent.y, wire_tangent.x))
    wire_position = path.positionAt(relative_position_on_wire)

    print(f"rotation is {wire_angle}")

    return (self.rotate(
        face_bottom_center, face_bottom_center + cq.Vector(0, 0, 1), wire_angle
    ).translate(wire_position - face_bottom_center), wire_angle)

cq.Face.faceOnWire = _faceOnWire

def textOnWire(txt: str, fontsize: float, distance: float, path: cq.Wire) -> cq.Solid:
    """Create 3D text with a baseline following the given path"""
    linear_faces = (
        cq.Workplane("XZ")
        .text(
            txt=txt,
            fontsize=fontsize,
            distance=distance,
            halign="left",
            valign="bottom",
        )
        .faces("<Y")
        .vals()
    )
    faces_on_path = [f.faceOnWire(path) for f in linear_faces]
    return cq.Compound.makeCompound(
        [cq.Solid.extrudeLinear(f[0], cq.Vector(0, 1, 0)) for f in faces_on_path]
    )

path = cq.Edge.makeThreePointArc(
    cq.Vector(0, 0, 0), cq.Vector(50, 30, 0), cq.Vector(100, 0, 0)
)
text_on_path = textOnWire(
    txt="The quick brown fox jumped over the lazy dog",
    fontsize=5,
    distance=1,
    path=path,
)
```
Do you have a hint? :)

jniclas, 2023-11-22:
Ah ok found it. Works now 👍🏻 
[image]
If someone wants to use it, thats the final source code:
```python
import cadquery as cq
from math import degrees, atan2

def _faceOnWire(self, path: cq.Wire) -> cq.Face:
    """Reposition a face from alignment to the x-axis to the provided path"""
    path_length = path.Length()

    bbox = self.BoundingBox()
    face_bottom_center = cq.Vector((bbox.xmin + bbox.xmax) / 2, 0, 0)
    relative_position_on_wire = face_bottom_center.x / path_length
    wire_tangent = path.tangentAt(relative_position_on_wire)
    wire_angle = degrees(atan2(wire_tangent.y, wire_tangent.x))
    wire_position = path.positionAt(relative_position_on_wire)

    print(f"rotation is {wire_angle}")

    return (self.rotate(
        face_bottom_center, face_bottom_center + cq.Vector(0, 0, 1), wire_angle
    ).translate(wire_position - face_bottom_center), wire_angle)

cq.Face.faceOnWire = _faceOnWire

def textOnWire(txt: str, fontsize: float, distance: float, path: cq.Wire) -> cq.Solid:
    """Create 3D text with a baseline following the given path"""
    linear_faces = (
        cq.Workplane("XZ")
        .text(
            txt=txt,
            fontsize=fontsize,
            distance=distance,
            halign="left",
            valign="bottom",
        )
        .faces("<Y")
        .vals()
    )
    faces_on_path = [f.faceOnWire(path) for f in linear_faces]
    return cq.Compound.makeCompound(
        [cq.Solid.extrudeLinear(f[0], f[0].normalAt(cq.Vector(0, 0, 0))) for f in faces_on_path] 
    )

r = 20 # Radius of the helix
p = 15 # Pitch of the helix
h = 30 # Height of the helix

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)

# Text
text_on_path = textOnWire(
    txt="The quick brown fox jumped over the lazy dog The quick brown fox jumped over the lazy dog",
    fontsize=5,
    distance=1,
    path=wire,
)
```

closed 2023-11-22 by jniclas