CadQuery/cadquery #1442: Knurling by cutting doesn't work with more complex cutting shapes

mryndzionek, 2023-11-12:
I'm trying to do knurling on a cylinder by cutting swept shapes. If I use a shape with a triangular section:
[image]
it works great:
[image]

With a rounded section:
[image]
it breaks down:
[image]
Any hints on how to fix this? I tried cutting in different order, unioning before cutting etc. and it always is wrong.

jmwright, 2023-11-13:
@mryndzionek Can you check `isValid()` on the rounded shapes before cutting and then after? https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=isvalid#cadquery.Shape.isValid

mryndzionek, 2023-11-13:
> @mryndzionek Can you check `isValid()` on the rounded shapes before cutting and then after? https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=isvalid#cadquery.Shape.isValid

I've checked `val().isValid()` and it only returns `True`.

mryndzionek, 2023-11-13:
I've noticed that I even have problem with unioning the two "directions". I can union same "directions":
[image]
but then when I try unioning those two together it fails. Similarly I can cut those two from a cylinder separately:
[image]
, but when I cut one after another the result is wrong:
[image]

jmwright, 2023-11-13:
You could try sweeping each individual cut (or all the cuts in one direction followed by all the cuts in the other direction) with sweep's `combine` argument set to `cut` to see if that gives better results.

https://github.com/CadQuery/cadquery/blob/7bc8c00dc1a81633daea4810e04762d61a2abd5a/cadquery/cq.py#L3175

jmwright, 2023-11-15:
@mryndzionek Are you able to post your code?

jmwright, 2023-11-15:
This issue could also be related. https://github.com/CadQuery/cadquery/issues/1437

jmwright, 2023-11-15:
I get CAD kernel errors if I make even small changes to your code, so I think you're running in to a kernel issue. I haven't found a magic work-around yet, and CQ-editor crashed my desktop environment on one try. Have you tried unioning all of the cuts together and then cutting the resulting compound at once from the cylinder?

mryndzionek, 2023-11-15:
> Have you tried unioning all of the cuts together and then cutting the resulting compound at once from the cylinder?

Yes, no major difference, maybe even somewhat worse result. For me CQ-editor deadlocks one some values (like small point radius, etc).

jmwright, 2023-11-15:
I tried your code in a development environment with OCP/OCCT 7.7.2 and got the same result. Have you tried varying the profile other than triangle vs rounded? For instance, a trapezoid and a trapezoid with rounded corners? I think you are stuck trying to find a combination that that CAD kernel can handle.

mryndzionek, 2023-11-15:
Trapezoid is also wrong:
[image]
The only profile which seems to work is a triangle.

jmwright, 2023-11-15:
Ok. I can only go up to 0.29 on the knurl cut_radius before the failure happens.
```python
res = knurl(res, height + 1, radius, 50, 0.29, 2, 100, 10)
```
There might be a clever way to use offset twist extrudes with union to achieve the same effect, but I really don't know.

lorenzncode, 2023-11-18:
Here is an example based on mailing list post:
https://groups.google.com/g/cadquery/c/So2xgLvxmXs/m/WzkdLhN1CAAJ

It doesn't resolve the OCCT issue though it happens to work for these values.
I'm sharing it more as a `Sketch` example.  Warning - the boolean ops are slow here with n=10.

```py
import cadquery as cq
import math

r = 20
h = 50
n = 10  # number of cuts (per direction);
a = 120
cut_angle = 50
cut_depth = 2
length = 2 * cut_depth
point_radius = 0.5

x1 = length * math.cos(math.radians(cut_angle / 2))
y1 = length * math.sin(math.radians(cut_angle / 2))

def doprofile(loc):
    s = (
        cq.Sketch()
        .push([(-cut_depth, 0)])
        .segment((0, 0), (x1, y1))
        .segment((x1, -y1))
        .close()
        .assemble()
        .reset()
        .vertices("<X")
        .fillet(point_radius)
        .reset()
    )
    return s.moved(loc)

s1 = cq.Sketch().parray(r, 0, 360, n).each(doprofile)

w = cq.Workplane()
w1 = w.placeSketch(s1).twistExtrude(h, a)
w2 = w.placeSketch(s1).twistExtrude(h, -a)

tool = w1.union(w2, clean=True, tol=1e-2)
part = w.circle(r).extrude(h).cut(tool, tol=1e-2)
```

[image]

mryndzionek, 2023-11-18:
I've tried using `twistExtrude` with the full circular profile and then combine with `intersect`. With `clean=True, tol=1e-2` it seems to be giving good results even for larger `point_radius`es:
[image]