CadQuery/cadquery #1445: Modelling half-ellipsoid that also produces a "nice" .step file

ray-pH, 2023-11-21:
I'm trying to model a half-ellipsoid, I'm not sure if there's a built-in primitive for it. Currently, what I'm doing is creating an ellipse arc, and then revolving it.
```python
import cadquery as cq

a = 10
b = 5
ell = cq.Workplane("XZ")\
    .moveTo(a,0)\
    .ellipseArc(a, b, 0, 90, startAtCurrent=True)\
    .lineTo(0,0)\
    .close().revolve(360, (0,0,0), (0,1,0))

```
This looks fine in cq, exactly what i want
[image]

But when i exported it into a .step file and opened it with FreeCad, it loads extremely slow and the resulting shape is not correct
[image]

I suppose this has something to do with #1338, based on that issue, the problem is related to degenerate point in the middle. The suggestion is to add some tolerance

```python
import cadquery as cq

a = 10
b = 5
eps=1e-3

ell = cq.Workplane("XZ")\
    .moveTo(a + eps,0)\
    .ellipseArc(a, b, 0, 90, startAtCurrent=True)\
    .lineTo(eps,0)\
    .close().revolve(360, (0,0,0), (0,1,0))
```
The .step file works fine with FreeCad, it loads super quickly
[image]

But now, there's a hole in the middle of it
[image]

My current workaround is just to patch the hole in the middle with a solid cylinder, is there a better way to do this?

lorenzncode, 2023-11-22:
Another way is to rotate only 180 degrees then mirror and union.

```py
import cadquery as cq

a = 10
b = 5

ell = (
    cq.Workplane("XZ")
    .moveTo(a, 0)
    .ellipseArc(a, b, 0, 90, startAtCurrent=True)
    .lineTo(0, 0)
    .close()
    .revolve(180, (0, 0, 0), (0, 1, 0))
)
ell = ell.mirror("XZ", union=True)
```

Here is the equivalent using Sketch:
```py
import cadquery as cq

a = 10
b = 5

ell = (
    cq.Workplane("XZ")
    .sketch()
    .ellipse(a, b)
    .push([(-a / 2, 0), (a / 2, -b)])
    .rect(a, 2 * b, mode="s")
    .finalize()
    .revolve(180, (0, 0, 0), (0, 1, 0))
    .mirror("XZ", union=True)
)
```

ray-pH, 2023-11-23:
That worked great, thanks

closed 2023-11-23 by ray-pH