CadQuery/cadquery #1449: [feature request] embed user selectable length units in step file outputs

greyltc, 2023-11-28:
The step files I write with cadquery always seem to have hard coded units of millimeter. The files always contain a line like:
```
#1870=( LENGTH_UNIT() NAMED_UNIT(*) SI_UNIT(.MILLI.,.METRE.) );
```
It would be a great new feature if we could pick from a list of units to write into that field to help with interoperability of the step files. Maybe the list to pick from could be something like: nanometers, micrometers, millimeters, meters, kilometers, inches, feet, miles

jmwright, 2023-11-28:
@greyltc Can you give an example of broken STEP interoperability with the current hard coded mm specification?

adam-urbanczyk, 2024-01-23:
Not currently implemented in CQ, but you can always use the OCCT API directly. E.g. setting units to um:
```python
import OCP

w = OCP.STEPControl.STEPControl_Writer()
OCP.Interface.Interface_Static.SetCVal_s("write.step.unit","UM")
```
NB: This seems to be a global operation in OCCT