CadQuery/cadquery #1462: Should recommend using explicit python version with conda, also miniconda

weskerfoot, 2023-12-14:
Thanks for making this tool, I had a couple issues installing with conda:

- Since it only works on specific python versions, I initially tried with whatever version I had in my environment, which is 3.11, which failed trying to install cq-editor, so I think the instructions should say to do something like `conda create -n cadquery python==3.10` since it's already clear that it won't work on >3.10 anyway.

- With a fresh anaconda environment it was taking an extremely long time to solve dependencies, miniconda worked almost immediately (I guess in general it's better to use miniconda since it doesn't come with all these extra dependencies)

adam-urbanczyk, 2023-12-20:
Are you sure that cadquery does not work with 3.11? Recent CI runs (before merging 3.12 support) were run with 3.11. IDK about the slowness, docs clearly suggest using [mambaforge](https://cadquery.readthedocs.io/en/latest/installation.html#install-the-conda-package-manager) and it works like a charm in CI and locally.

weskerfoot, 2023-12-21:
> Are you sure that cadquery does not work with 3.11? Recent CI runs (before merging 3.12 support) were run with 3.11. IDK about the slowness, docs clearly suggest using [...]

That's fair, I missed the part about using mambaforge I guess. As for python 3.11, I would have to try again with a clean environment to see what the error I got was. I'll give that a try and see if it happens again.  The reason I thought it wasn't supported is because it mentions only >=3.8,<3.10 are supported [here](https://cadquery.readthedocs.io/en/latest/installation.html#install-via-pip). Maybe that needs to be updated if it actually works.

jmwright, 2024-01-16:
A [PR is open](https://github.com/CadQuery/cadquery/pull/1498) to fix the pip documentation issue, and numerous community members are running Python 3.11 with both conda and pip without issue. Closing. Thanks @weskerfoot

closed 2024-01-16 by jmwright