CadQuery/cadquery #1463: How to extrude a loft?

ninja-, 2023-12-15:
I believe I tried every combination of commands, especially around .wires(), .faces() .toPending() etc. however I am struggling to extrude a loft. Can you help me add 3D depth to the created loft?
Thanks :)

```python
result = (
    cq.Workplane("XZ")
    .ellipseArc(90, 70, angle1=0, angle2=30,makeWire=True)
    .moveTo(10, 0)
    .ellipseArc(90, 70+10/2, angle1=0, angle2=30, makeWire=True)
    .loft()
 )
```

[image]

ninja-, 2023-12-16:
I finally found a working hack here: https://github.com/CadQuery/cadquery/issues/884#issuecomment-918121609

is there possibly a way to do this with the fluent api?

adam-urbanczyk, 2024-01-02:
AFAICT there is currently no clean way in the fluent API.

adam-urbanczyk, 2024-01-03:
This works, but is not exactly pretty:
```python
result = (
    cq.Workplane("XZ")
    .ellipseArc(90, 70, angle1=0, angle2=30,makeWire=True)
    .moveTo(10, 0)
    .ellipseArc(90, 70+10/2, angle1=0, angle2=30, makeWire=True)
    .loft()
    .faces(cq.selectors.AreaNthSelector(-1))
    .wires()
    .toPending()
    .extrude(10, combine=False)
 )
```
`loft()` seems to result in a incorrect solid, and that is why the additional gymnastics are needed.

lorenzncode, 2024-06-09:
Here is an example using the new free function API.

```py
import cadquery as cq
from cadquery.occ_impl.shapes import extrude, loft

# define the two wires with Workplane
wp = (
    cq.Workplane("XZ")
    .ellipseArc(90, 70, angle1=0, angle2=30, makeWire=True)
    .tag("wi1")
    .moveTo(10, 0)
    .ellipseArc(90, 70 + 10 / 2, angle1=0, angle2=30, makeWire=True)
    .tag("wi2")
)

# apply loft to wires to create face (free function API)
f = loft(wp.wires(tag="wi1").val(), wp.wires(tag="wi2").val())

# extrude face
result = extrude(f, (0, -10, 0))

# or extrude face given normal already defined by Workplane
result2 = extrude(f, wp.plane.zDir)
```