CadQuery/cadquery #1471: Feedback on Latest OCP Wheels

jmwright, 2023-12-29:
I am looking for feedback on the OCP wheels, but you will need to install the wheel directly rather than through PyPI for now. I want to test the wheels before I push them to PyPI. Please test the wheels [here](https://github.com/CadQuery/ocp-build-system/actions/runs/7353258115). The Python 3.10 and 3.12 wheels worked fine for me on Ubuntu 22.04. However, the nlopt package seems to have a problem with Python 3.12, so it may not possible to get a full working CadQuery environment with that Python version. There was a problem with nlopt soon after the release of Python 3.11, so the same thing may be happening here. I don't see an issue on the nlopt tracker about Python 3.12 yet though.

Mac M1 and Linux ARM64 wheels are not included yet, but there is a PR open on the [ocp-build-system repo](https://github.com/CadQuery/ocp-build-system) that may enable those to be made available at some point.

lorenzncode, 2023-12-31:
I tested the non-macos wheels locally.

pytest results (linux Fedora 38, windows 10 VM):
Python3.8, Python3.9: two known test failures #1441
Python3.12: fail

`cadquery.vis`:
No issue found on linux Python3.8 through 3.11. 
I tested on a real Windows 11 machine (only with Python3.11) - no issue.

Logs with pip list and pytest output attached.
[pip_logs.zip](https://github.com/CadQuery/cadquery/files/13802085/pip_logs.zip)

jmwright, 2024-01-01:
@lorenzncode So there are no new/unexpected errors, is that correct? Do these OCP wheels seem ready to publish from your perspective?

lorenzncode, 2024-01-01:
> @lorenzncode So there are no new/unexpected errors, is that correct? Do these OCP wheels seem ready to publish from your perspective?

Correct, no new/unexpected errors in my testing.  Yes, it looks ready to publish to me.

jmwright, 2024-01-02:
Thanks @lorenzncode . The OCP 7.7.2 wheels have been published for non-ARM Linux and MacOS. ARM variants are in progress and will hopefully be published at a later time.

closed 2024-01-02 by jmwright