CadQuery/cadquery #1473: What does question marks(?)(in assebly's constrain function) mean?

huskier, 2024-01-01:
In document's PointInPlane example, there is the following code. What does question marks(?)(in assebly's constrain function) mean? Also, where could we find the meanings of inner_horiz and inner_vert?

```
# constrain the bottom of the box to be on the plane defined by inner_horiz:
assy.constrain("box@faces@<Z", "bracket?inner_horiz", "PointInPlane")
# constrain the side of the box to be 0.2 units from the plane defined by inner_vert
assy.constrain("box@faces@<Y", "bracket?inner_vert", "PointInPlane", param=0.2)
```

adam-urbanczyk, 2024-01-01:
`?name` refers to the tagged item `name`. This should be stated explicitly in the docs.

huskier, 2024-01-02:
Thank you @adam-urbanczyk.
And then "bracket?inner_horiz" means "bracket's ?inner_horiz" literally?

adam-urbanczyk, 2024-01-02:
Indeed. Obviously it assumes that `inner_horiz` was tagged.

closed 2024-01-02 by huskier