CadQuery/cadquery #1475: Python 3.12 Support When Installing via pip

jmwright, 2024-01-02:
I am posting this here as a known issue that we can point to when people ask.

As mentioned in [this issue](https://github.com/CadQuery/cadquery/issues/1471), our Python 3.12 support is broken right now when isntalling via pip because of our dependency `nlopt`. I expect this package will be fixed for Python 3.12 eventually (just like it was with 3.11), but for now we have to wait.

CadQuery installed via conda ([instructions here](https://cadquery.readthedocs.io/en/latest/installation.html#install-the-conda-package-manager)) seems to work fine with Python 3.12, so that is an option.

Jfdezmtz, 2024-02-05:
Hello, was wondering if there was any update on this? Would love to use this for a project, but we are running 3.12.

jmwright, 2024-02-05:
@Jfdezmtz You can try installing CadQuery, but nlopt still throws an error via pip for me on Python 3.12 in Linux. As far as I can tell, that is the only broken dependency, but it is not one that can be left out.

Jfdezmtz, 2024-02-05:
@jmwright Yes, nlopt still throws me an error. I was wondering if there was any update or ETA on when this issue might be fixed?

jmwright, 2024-02-05:
No ETA. That dependency is not under the CadQuery umbrella. 

https://github.com/stevengj/nlopt/issues/548