CadQuery/cadquery #1479: Is it possible to set color to a 3D object without using an assembly?

yuxguo, 2024-01-05:
I want to set color to the 3D object via the codes like the following:
```python
import cadquery as cq

box_lower = cq.Workplane().pushPoints([(0, 0), ]).rect(3, 3).extrude(1) # with blue color
box_upper = box_lower.faces(">Z").workplane().pushPoints([(0, 0), ]).rect(1, 1).extrude(1) # with gree color

display(box_upper)
```
I don't want to use the assembly feature because that would result in too many parts.
Is it possible to set color to a 3D object without using an assembly?

jmwright, 2024-01-05:
@yuxguo What execution environment are you using? I see `display` in your code above, so it's not CQ-editor. When using CQ-editor, you can set the part color when you call `show_object`, but that information will not be preserved when you export the object to STEP. I don't think it's currently possible to export with colors to STEP without an assembly. You can do a fused STEP export with assembly that will preserve colors if you want to end up with a single object.

yuxguo, 2024-01-06:
Thanks for your reply; I use the IPython `display` to display the 3D objects.

closed 2024-01-07 by yuxguo