CadQuery/cadquery #1480: wires().toPending().extrude()

bragostin, 2024-01-05:
This code 
```python
import cadquery as cq
from math import cos, sin, tan, radians
import numpy as np

(L, W, H) = (200., 100., 100.)
nozzle_length = 20.
nozzle_diameter = 5.
nozzle_thickness = 1.
angle = 45.
dx = ( nozzle_diameter / 2 + nozzle_thickness ) * tan( radians(angle) )
dz = 1.

box = cq.Workplane("XY").box(L, W, H)
box = box.shell(nozzle_thickness, kind='intersection')
box = box.faces("<X").workplane(offset=dx).center(-W/4, 0).transformed((0, -angle, 0)).circle(nozzle_diameter/2).circle(nozzle_diameter/2+2*nozzle_thickness).extrude(-nozzle_length-dx)
box = box.faces("<X[-2]").workplane(offset=-dx).center(-W/2, 0).transformed((0, angle, 0)).circle(nozzle_diameter/2).circle(nozzle_diameter/2+2*nozzle_thickness).extrude(nozzle_length+dx)

fluid = cq.Workplane("XY").box(L, W, H)
fluid = fluid.cut(box)

faces = fluid.faces("<Z").workplane().transformed((0, 0, 0)).section(-H/2).faces()
f = faces.val()
ctw = faces.wires().val().Center().toTuple() # Retrieve the center coordinates of the face
nV = np.array( f.normalAt(cq.Vector(ctw)).toTuple()) # Retrieve the face's normal vector
nV = nV / np.linalg.norm(nV) * dz # Normalize normal vector
fluid = cq.Workplane('XY') # Define workplane
fluid = fluid.add( cq.Solid.extrudeLinear(f, tuple(nV)) )
```
produces the intended result (image attached), while this one
```python
import cadquery as cq
from math import cos, sin, tan, radians
import numpy as np

(L, W, H) = (200., 100., 100.)
nozzle_length = 20.
nozzle_diameter = 5.
nozzle_thickness = 1.
angle = 45.
dx = ( nozzle_diameter / 2 + nozzle_thickness ) * tan( radians(angle) )
dz = 1.

box = cq.Workplane("XY").box(L, W, H)
box = box.shell(nozzle_thickness, kind='intersection')
box = box.faces("<X").workplane(offset=dx).center(-W/4, 0).transformed((0, -angle, 0)).circle(nozzle_diameter/2).circle(nozzle_diameter/2+2*nozzle_thickness).extrude(-nozzle_length-dx)
box = box.faces("<X[-2]").workplane(offset=-dx).center(-W/2, 0).transformed((0, angle, 0)).circle(nozzle_diameter/2).circle(nozzle_diameter/2+2*nozzle_thickness).extrude(nozzle_length+dx)

fluid = cq.Workplane("XY").box(L, W, H)
fluid = fluid.cut(box)

fluid = fluid.faces("<Z").workplane().section(-H/2).wires().toPending().extrude(dz)
```
yields a `Null TopoDS_Shape Object`. Is it to be expected?

I know there are simpler ways to generate this object, but I need to be able to switch between 3D and 2D.
[image]

lorenzncode, 2024-01-07:
To produce the same result with the second example change the final extrude call to specify `combine=False`.

> yields a Null TopoDS_Shape Object. Is it to be expected?

I think it may not be expected that `findSolid` returns a Compound without a Solid.   

As in this example, `findSolid` returns a Compound with Face and fuse then results in the Null object:

```py
import cadquery as cq

box = cq.Workplane().box(1, 2, 3)
face_compound = cq.Compound.makeCompound(box.faces(">Z").val())

res = (
    cq.Workplane()
    .box(1, 2, 3)
    .add(face_compound)  # no purpose other than to break extrude combine=True
    .faces(">Z")
    .workplane()
    .circle(0.2)
    .extrude(1)
)
```

bragostin, 2024-01-07:
This works, thank you!

closed 2024-01-07 by bragostin