CadQuery/cadquery #1481: XML In ReadTheDocs Documentation

jmwright, 2024-01-05:
@lorenzncode @adam-urbanczyk Do we know why XML is showing up in our online documentation? I have tried in both Brave and Firefox with the same results. I did a search for the term `color` when answering a question this morning, and the following is what I saw in the search results.

[image]

sethfischer, 2024-01-05:
The Sphinx `cadquery` directive (`cadquery/cq_directive.py`) generates a VTK XML file which is embedded in a JavaScript variable:

```html
<div class="cq-vtk" style="text-align:left;float:left;border: 1px solid #ddd; width:100%; height:500px">
    <script>
        var parent_element = document.currentScript.parentNode;
        var data = [{"shape": "<?xml version=\"1.0\"?>\n<VTKFile type=\"PolyData\"...
        render(data, parent_element);
    </script>
</div>
```

Sphinx search is indexing the contents of the `<script>` tag.

jmwright, 2024-01-16:
@sethfischer Any ideas on how to fix this?

sethfischer, 2024-01-17:
I haven't yet found a solution. There doesn't appear to be any relevant [configuration options](https://readthedocs-sphinx-search.readthedocs.io/en/latest/configuration.html). It's on my todo list for sethfischer/sphinxcontrib-cadquery. I'll update here as I make progress.

lorenzncode, 2026-09-06:
Fixed externally in https://github.com/readthedocs/readthedocs.org/pull/12757 !

closed 2026-09-06 by lorenzncode