CadQuery/cadquery #1487: Different workplane origin position after cut solid and after cutBlind

sadr0b0t, 2024-01-08:
Hello,

I am trying to cut some figure from base figure horizontally and then make some drawings on a vertical surface of the same base figure and I have noted, that when I take vertical workplane of the solid, horizontal position of the center is different, depending on how I have made a cut - either with cutBlind or cutting extruded solid of the same shape.

My simplified code:
~~~python
box1 = (
    cq.Workplane("XY")
    .rect(6, 10, centered=(False, True))
    .extrude(4)
)

box1_cutBlind = (
    box1.faces("<X").workplane(offset=-1)
    .rect(3, 4)
    .cutBlind(-4)
)

box1_cutSolid = box1.cut(
    cq.Workplane().copyWorkplane(
        box1.faces("<X").workplane(offset=-1))
    .rect(3, 4)
    .extrude(-4)
)

#box1 = box1_cutBlind
box1 = box1_cutSolid

box2_v1 = (
    box1.faces("<X").workplane()
    .rect(3, 4)
)

box2_v2 = (
    box1.faces(">Z").workplane()
    .rect(4, 3, centered=(False, True))
)

#show_object(box1_cutBlind)
#show_object(box1_cutSolid)
show_object(box1)
#show_object(box2_v1)
show_object(box2_v2)
~~~

box1: either box1_cutBlind and box1_cutSolid visually give the same result. "offset" param tunes horizontal position of the cut solid. Then box2_v2 draws a rectangle on the top face of the original box1, one of the edges of the rect is placed on the 0 position of X.

So, tuning 'offset' param would move horizontal position of the cut solid on axes X. With box1_cutBlind this would also move topmost rectangle along the X axes. With box1_cutSolid option the topmost rectangle would always stay on the 0 position on axes X.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

jmwright, 2024-01-09:
I often use the form `.workplane(centerOption="CenterOfBoundBox")` to avoid issues like this.

sadr0b0t, 2024-01-09:
yes, using centerOption="CenterOfBoundBox" gives same result both for cutBlind and cut solid
~~~python
box2_v3 = (
    box1.faces(">Z").workplane(centerOption="CenterOfBoundBox")
    #box1.faces(">Z").workplane(centerOption="CenterOfMass")
    #box1.faces(">Z").workplane(centerOption="ProjectedOrigin")
    #box1.faces(">Z").workplane()
    .rect(4, 3, centered=(False, True))
)
~~

jmwright, 2024-01-17:
@sadr0b0t  I don't think I understand. The following two snippets produce the same result for me.

## Boolean Cut
```python
import cadquery as cq

box1 = (
    cq.Workplane("XY")
    .rect(6, 10, centered=(False, True))
    .extrude(4)
)

# Boolean operation
box1 = box1.cut(
    cq.Workplane().copyWorkplane(
        box1.faces("<X").workplane(offset=-1))
    .rect(3, 4)
    .extrude(-4)
)

# Blind cut operation
# box1 = (box1.faces("<X").workplane(offset=-1)
#     .rect(3, 4)
#     .cutBlind(-4))

box2_v2 = (
    box1.faces(">Z").workplane(centerOption="CenterOfBoundBox")
    .rect(4, 3, centered=(True, True))
)

show_object(box1, options=dict(alpha=0.5))
show_object(box2_v2)
```
[image]

## Blind Cut
```python
import cadquery as cq

box1 = (
    cq.Workplane("XY")
    .rect(6, 10, centered=(False, True))
    .extrude(4)
)

# Boolean operation
# box1 = box1.cut(
#     cq.Workplane().copyWorkplane(
#         box1.faces("<X").workplane(offset=-1))
#     .rect(3, 4)
#     .extrude(-4)
# )

# Blind cut operation
box1 = (box1.faces("<X").workplane(offset=-1)
    .rect(3, 4)
    .cutBlind(-4))

box2_v2 = (
    box1.faces(">Z").workplane(centerOption="CenterOfBoundBox")
    .rect(4, 3, centered=(True, True))
)

show_object(box1, options=dict(alpha=0.5))
show_object(box2_v2)
```
[image]

You'll have to move your 2D rectangle manually to match the position of the cut.
```python
import cadquery as cq

box1 = (
    cq.Workplane("XY")
    .rect(6, 10, centered=(False, True))
    .extrude(4)
)

# Boolean operation
# box1 = box1.cut(
#     cq.Workplane().copyWorkplane(
#         box1.faces("<X").workplane(offset=-1))
#     .rect(3, 4)
#     .extrude(-4)
# )

# Blind cut operation
box1 = (box1.faces("<X").workplane(offset=0)
    .rect(3, 4)
    .cutBlind(-4))

box2_v2 = (
    box1.faces(">Z").workplane(centerOption="CenterOfBoundBox")
    .move(-1, 0).rect(4, 3, centered=(True, True))
)

show_object(box1, options=dict(alpha=0.5))
show_object(box2_v2)
```