CadQuery/cadquery #1488: Simple combination of some ellipses and extrude eats memory and hangs app

sadr0b0t, 2024-01-09:
Hello,

This simple code being ran from CQ-Editor eats about 2Gb of memory (from 3.4 to 5.4Gb), cpu 100% and hangs app for about a minute or so. The app did terminate when I first saw that, but it was the os who killed it because of memory consumption. If I give enough memory, it would finally render the model after about a minute.

~~~python
import cadquery as cq
result2 = (
    cq.Workplane("XY")
    .ellipse(30, 25).ellipse(27, 22).extrude(6)
    .ellipse(30, 25).extrude(6)
)
~~~

Just paste code to CQ-Editor and press render. It happens all of the time.

[image]
[image]
[image]

I have made some experiments and it seems that this is some kind of lucky magic combination or figures, operations and numbers. For example, it won't hang in this way:

~~~python
result10 = (
    cq.Workplane("XY")
    .ellipse(10,10).ellipse(8,8).extrude(6)
    .ellipse(10,10).extrude(6)
)
~~~

also won't hang in this ways:
~~~
result3 = (
    cq.Workplane("XY")
    .ellipse(30, 25).ellipse(27, 22).extrude(6)
    .ellipse(30, 26).extrude(6)
)

result5 = (
    cq.Workplane("XY")
    .ellipse(30, 25).extrude(6)
    .ellipse(30, 25).extrude(6)
)

result5 = (
    cq.Workplane("XY")
    .ellipse(10, 25).ellipse(27, 22).extrude(6)
    .ellipse(10, 25).extrude(6)
)

result9 = (
    cq.Workplane("XY")
    .ellipse(26, 25).ellipse(27, 22).extrude(6)
    .ellipse(26, 25).extrude(6)
)
~~~

it would also hang in this way but for much less time:
~~~python
result4 = (
    cq.Workplane("XY")
    .ellipse(30, 25).ellipse(27, 22)
    .ellipse(30, 25).extrude(6)
)
~~~

even less time to hang:
~~~python
result6 = (
    cq.Workplane("XY")
    .ellipse(31, 25).ellipse(27, 22).extrude(6)
    .ellipse(31, 25).extrude(6)
)
~~~

won't hang, but would draw nothing:
~~~python
result7 = (
    cq.Workplane("XY")
    .ellipse(27, 25).ellipse(27, 22).extrude(6)
    .ellipse(27, 25).extrude(6)
)
~~~

lorenzncode, 2024-01-10:
Try relaxing the 3D Viewer tolerances:

[image]

See also duplicate: https://github.com/CadQuery/CQ-editor/issues/384

sadr0b0t, 2024-01-13:
> BTW you probably wanted:

yes, I have already drawn what I wanted in similar way, but one can press "render" when design syntax "compiles", but has such logical errors, and receive hang or killed app.