CadQuery/cadquery #1489: Linux aarch64 (ARM64) Python Wheels

jmwright, 2024-01-10:
See [the announcement](https://github.com/CadQuery/ocp-build-system/issues/20) on the ocp-build-system repo. You will need to download the cadquery-ocp wheel for your version of Python, and do a local install. In that shared directory there is also a requirements.txt file that contains the CadQuery dependencies so that you can try to hack together a working CadQuery install. However, there are some challenges.

* Tools for building software (gcc, cmake, build-essential, etc) have to be installed for installing/building nlopt.
* nlopt fails to build for me without reporting why

If someone in the community can help figure out the issue with nlopt compilation, that would be great.

If you are interested, please hack around and let me know what you find. Use `pip install -r requirements.txt` to install the dependencies in your Python environment after doing a local install of the cadquery-ocp wheel.

hf-krechan, 2024-02-09:
Hey @jmwright I tried to use the [`cqgridfinity`](https://github.com/michaelgale/cq-gridfinity) python package which needs `cadquery`.

After I successfully installed the arm64 cad query wheel from https://github.com/jdegenstein/ocp-build-system/releases/tag/7.7.2_macos_arm64 I stumbled over the same issue with `nlopt`.

On my search I found [this issue](https://github.com/DanielBok/nlopt-python/issues/13).
There are three other issues linked, which got all closed.
I hoped to find a solution there, but all issues with `nlopt` got "solved" by replacing it with `scipy`.

I gave it a try, but I was not successful so far.

The question I am asking myself now, what kind of approach would be more promising: Either to replace `nlopt` with `scipy` or try to contribute to `nlopt` and make it work on arm64?

There is already a two year old [issue which mentions the problem with the M1 processors](https://github.com/DanielBok/nlopt-python/issues/13).

thasti, 2024-02-10:
I also succesfully tested those, thanks for providing those wheels. nlopt didn't act up for me, installed just fine out of the box via pip (Python 3.12.1).

hf-krechan, 2024-02-11:
I used also Python 3.12.1 but on my Macbook Pro with M1 processor.
But if I want to install `nlopt` with 
```bash
pip install nlopt
```

I get the error message
```bash
ERROR: Could not build wheels for nlopt, which is required to install pyproject.toml-based projects
```

I found [this comment](https://github.com/stevengj/nlopt/issues/439#issuecomment-1595191483) which deactivates some options to compile `nlopt`:

> I am able to run on an M2 MacBook Pro by following the NLopt steps (although slightly modified):
> 
> 1. Open terminal and cd to desired directory (e.g. Documents).
> 2. git clone https://github.com/stevengj/nlopt.git
> 3. cd nlopt
> 4. mkdir build
> 5. cd build
> 6. cmake .. -DNLOPT_OCTAVE=Off -DNLOPT_MATLAB=Off -DNLOPT_GUILE=Off
> 7. make
> 8. sudo make install
> 
> If you need a CMakeLists.txt template please refer to my comment on #502.

@thasti do you also used an arm architecture?

thasti, 2024-02-11:
Hi. Indeed, I'm on an M2 Air running Fedora Asahi Remix. nlopt simply installs from wheels for me (i.e. is not compiled locally), no issues:

```
(venv_nlopt) [thasti@slab tmp]$ pip install nlopt
Collecting nlopt
  Using cached nlopt-2.6.2-cp312-cp312-linux_aarch64.whl
Collecting numpy>=1.14 (from nlopt)
  Obtaining dependency information for numpy>=1.14 from https://files.pythonhosted.org/packages/79/f8/97f10e6755e2a7d027ca783f63044d5b1bc1ae7acb12afe6a9b4286eac17/numpy-1.26.4-cp312-cp312-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata
  Downloading numpy-1.26.4-cp312-cp312-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.3/62.3 kB 1.9 MB/s eta 0:00:00
Downloading numpy-1.26.4-cp312-cp312-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (13.9 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.9/13.9 MB 33.7 MB/s eta 0:00:00
Installing collected packages: numpy, nlopt
Successfully installed nlopt-2.6.2 numpy-1.26.4
```

Nonetheless, I tried and I also have no troubles following the step-by-step instructions above without disabling anything (i.e. I just run `cmake ..` instead of `cmake .. -DNLOPT_OCTAVE=Off -DNLOPT_MATLAB=Off -DNLOPT_GUILE=Off`) and it builds just fine.

hf-krechan, 2024-02-11:
Alright, that is good to know. Thanks for testing :)

I tested now the the commands to build (is this the right term? ^^') `nlopt`.
These commands worked for me, but if I try to install `nlopt` in my virtual environment with `pip install nlopt` it crashes again with this error

```bash
$ (.venv)  pip install nlopt
Collecting nlopt
  Using cached nlopt-2.6.2.tar.gz (2.0 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy>=1.14 in ./.venv/lib/python3.12/site-packages (from nlopt) (1.26.4)
Building wheels for collected packages: nlopt
  Building wheel for nlopt (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for nlopt (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [261 lines of output]
      running bdist_wheel
      running build
      running build_ext
      cmake version 3.28.3
```

These are only the first lines.
In the collapsed section you can find the whole output.

<details>

<summary>Whole error message </summary>

```bash
Collecting nlopt
  Using cached nlopt-2.6.2.tar.gz (2.0 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy>=1.14 in ./.venv/lib/python3.12/site-packages (from nlopt) (1.26.4)
Building wheels for collected packages: nlopt
  Building wheel for nlopt (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for nlopt (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [261 lines of output]
      running bdist_wheel
      running build
      running build_ext
      cmake version 3.28.3
      
      CMake suite maintained and supported by Kitware (kitware.com/cmake).
      CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
        Compatibility with CMake < 3.5 will be removed from a future version of
        CMake.
      
        Update the VERSION argument <min> value or use a ...<max> suffix to tell
        CMake that the project does not need compatibility with older versions.
      
      
      -- The C compiler identification is AppleClang 15.0.0.15000100
      -- The CXX compiler identification is AppleClang 15.0.0.15000100
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
[... 224 lines cut ...]
          self.distribution.run_command(command)
        File "/private/var/folders/5_/rkzx1ly96xbcb33y2phnbq940000gn/T/pip-build-env-d_uajgbu/overlay/lib/python3.12/site-packages/setuptools/dist.py", line 963, in run_command
          super().run_command(command)
        File "/private/var/folders/5_/rkzx1ly96xbcb33y2phnbq940000gn/T/pip-build-env-d_uajgbu/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/private/var/folders/5_/rkzx1ly96xbcb33y2phnbq940000gn/T/pip-build-env-d_uajgbu/overlay/lib/python3.12/site-packages/setuptools/_distutils/command/build.py", line 131, in run
          self.run_command(cmd_name)
        File "/private/var/folders/5_/rkzx1ly96xbcb33y2phnbq940000gn/T/pip-build-env-d_uajgbu/overlay/lib/python3.12/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/private/var/folders/5_/rkzx1ly96xbcb33y2phnbq940000gn/T/pip-build-env-d_uajgbu/overlay/lib/python3.12/site-packages/setuptools/dist.py", line 963, in run_command
          super().run_command(command)
        File "/private/var/folders/5_/rkzx1ly96xbcb33y2phnbq940000gn/T/pip-build-env-d_uajgbu/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "<string>", line 28, in run
        File "<string>", line 70, in build_extension
      StopIteration
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for nlopt
Failed to build nlopt
ERROR: Could not build wheels for nlopt, which is required to install pyproject.toml-based projects
```

</details>

adam-urbanczyk, 2024-02-11:
Feel free to continue the discussion here, but it might be more efficient to ask on the nlopt issue tracker.

thasti, 2024-02-11:
Not obvious to me what is going wrong with your build. I would also suggest you take your problem to the nlopt issue tracker (https://github.com/stevengj/nlopt/issues) since this seems unrelated to Linux aarch64 discussed here, people are more likely to be able to help you there. 

@adam-urbanczyk let me know if any other tests would be useful for you. Can you confirm the issue you had with nlopt on linux aarch64 has been resolved in the meantime or do you still see failures?

jmwright, 2024-02-11:
I have not been able to get around the Python 3.12 problem on Debían 12 aarch64 running on a Pinebook Pro, and I also tried on a Rock Pro 64 running the same distro. However, I have not tried with the different cmake options listed above. @hf-krechan Is it possible that the proper nlopt and nlopt-dev packages are not installed via your distro's package manager that would allow nlopt to build locally with the new cmake options? If that is not the remaining issue, I think the nlopt issue tracker is the best place for this (as mentioned above).

hf-krechan, 2024-02-11:
I did not know that you can also install nlopt with a package manager.
On macOS I use homebrew and there is a `nlopt` package: https://formulae.brew.sh/formula/nlopt#default

But the `pip install nlopt` command still fails.
I will move to the nlopt issue tracker with this problem.

Do you have some opinions about the idea to replace `nlopt` with `scipy`?
As far as I see it get's only used in the `sketch_solver.py` module.

jmwright, 2024-02-11:
> Do you have some opinions about the idea to replace nlopt with scipy?

@adam-urbanczyk Will probably have better insight on that. I am fine sticking with nlopt and contributing to that package unless there are other compelling reasons to switch.

adam-urbanczyk, 2024-02-11:
Definitely not scipy. Long term we should switch to casadi (i.e. what is used by the assy solver) or get rid of the sketch solver completely.

adam-urbanczyk, 2024-02-11:
I don't understand what precisely are we discussing here. It does not look like a nlopt issue per se. It uses CMake and not pip for building... Maybe try the offical way first. 

BTW conda-forge seems to have a aarch64/arm64 nlopt pacakges also for py3.12.

hf-krechan, 2024-02-11:
I thought the topic would be the issue to install `cadquery` on an ARM64 machines.

I am not familiar with conda and I am a little bit afraid to crash my Python setup which I need for work.
I will try find a way to install the `nlopt-python` on macOS.

Thanks for all you inputs ☺️

adam-urbanczyk, 2024-02-11:
FYI mamba/conda is the best supported way to install cadquery, install to a local dir if you are afraid of side effects: https://cadquery.readthedocs.io/en/latest/installation.html#install-the-conda-package-manager

hf-krechan, 2024-02-11:
Just in case if someone like does also not know about `miniforge`.
Miniforge is a minimal installer for the Conda package manager. And with this you can install `nlopt` on a M1 machine.

## Install Miniforge 

You can install it with
```bash
brew install --cask miniforge
```

After the installation you need to run
```bash
conda init "$(basename "${SHELL}")"
```

## Create a Conda Environment

Instead of using `venv` you use `conda` to create a virtual environment.
```bash
conda create --name myenv python=3.11
```
Python 3.11 is the latest version at the moment.
I guess it is necessary that you have Python 3.11 installed already.

Activate the conda environment with

```bash
conda activate myenv
```

Now you can install `nlopt` with

```bash
conda install nlopt
```

If you need a package which is not available in the conda repositories, you can still use `pip`.
But be careful with this approach, it can easily lead to version conflicts.

hf-krechan, 2024-02-11:
Thanks for this hint.
I tried it and it worked :)
And I learned a lot about Anaconda, Conda, Miniforge and Mamba 😄

jmwright, 2024-02-19:
Linux aarch64 and MacOS ARM64 wheels have been attached to the ocp-build-system release [here](https://github.com/CadQuery/ocp-build-system/releases/tag/7.7.2.0). It is possible to do a pip install of the wheels on that release by copying the URL for the file that matches your OS and Python version and then doing `pip install URL_OF_CORRECT_WHEEL_FILE`.

RaubCamaioni, 2025-08-09:
Ran into issues building nlopt using the packages provided from PyPI.
The errors are easier to read when installing from nlopt GitHub source code: https://github.com/DanielBok/nlopt-python.
I was able to build on an aarch64 system with the following steps.
```
# activate your virtual environment
sudo apt install swig # swig required, causes missing file error
git clone https://github.com/DanielBok/nlopt-python
cd nlopt-python
pip install .
```