CadQuery/cadquery #1491: Parameter names for box/rect width and height are confusing

BlueDrink9, 2024-01-11:
I recognise this may be subjective, and there may be compatibility issues with altering the names that prevent change here. I wanted to open a discussion and see if I'm alone in getting these constantly mixed up.

Currently width is used for distance in Y, length for X. The problem is, depending on the shape, these might be swapped. For example, the front of a car is its width, which would be on the X. From quick googling, the only consensus I can find is that length is the long side and width is the short side, which is clearly not appropriate for cad modelling.

Openscad uses width for X as well, but uses depth for Y - which is much less ambiguous.

I propose renaming width to depth in this instance, but really just want to discuss and see what the maintainers think

BlueDrink9, 2024-01-11:
Alternatively, `x_length` etc would be the least ambiguous of all

adam-urbanczyk, 2024-01-12:
The docstring is quite explicit.

BlueDrink9, 2024-01-12:
Yes, the docstring is. But editors don't always show that. Sometimes we just get hints for the parameter names, for example autocomplete.

Even if you do have the docstring hover over the text, having to read a sentence on the fly is more annoying than reading a single parameter. If you aren't using cq on a regular basis, a user very possibly does need to check which parameter is which every time they use it.

In these situations, it would make cq tidier to have more expressive parameter names