CadQuery/cadquery #1497: CadQuery 2.4.0**  does not exist (perhaps a typo or a missing channel).

huskier, 2024-01-16:
My WSL2(UBUNTU22.04.3) has cadquery 2.3.1 installed. When I try to intall the latest version through "mamba install cadquery", the error occurs which reads "cadquery 2.4.0**  does not exist (perhaps a typo or a missing channel)." The command "mamba install cadquery=2.4.0" gives the same error.

I wonder whether the CadQuery 2.4.0 is ready for mamba install?

jmwright, 2024-01-16:
@adam-urbanczyk I followed [the release documentation](https://github.com/CadQuery/cadquery/wiki/Developer-Doc:-Creating-a-New-Release) that has worked for previous releases, so I am not sure why this would be broken. Any ideas?

jmwright, 2024-01-16:
This install line works to install 2.4.0 from conda.
```
conda install cadquery::cadquery
```

huskier, 2024-01-17:
Thanks. @jmwright @adam-urbanczyk

closed 2024-01-17 by huskier