CadQuery/cadquery #1499: About reading the source code.

huskier, 2024-01-16:
I want to dive into the source code of CadQuery. Are there any auxiliary documents or texts to consult to grasp the holoview? 

I've had read through the "cadquery.readthedocs.io", and I think it's for application, not for developing the code.

Back to 2015, I had contributed some code. However, it's a long time ago, and I've almost forgotten the context of the source code. Meanwhile, there is a huge change since then.

Any suggestions are appreciate!

jmwright, 2024-01-16:
If you are wanting to work at a low (OCCT) level on CQ, you may want to be familiar with OpenCASCADE's API ([more info on the layers](https://cadquery.readthedocs.io/en/latest/primer.html#cadquery-api-layers)). At a higher level, there is a part of the docs that you have probably already seen [here](https://cadquery.readthedocs.io/en/latest/primer.html#). Quite a bit of the code has decent docstrings, and the tests are useful when trying to figure out how things work. I still refer back to them periodically.

michaelgale, 2024-01-16:
Absolutely agree that the docstrings are the best place to start.  In particular the `Workplane` class (in `cq.py`).  Then, I would check out the `occ_impl/shapes.py` module. It shows how the underlying OCCT objects are glued to objects held by a `Workplane`.  Lastly, the `selectors.py` is useful, if not interesting to see how powerful object filtering can be for building complex geometry.

huskier, 2024-01-18:
Thanks. @jmwright @michaelgale

closed 2024-01-18 by huskier