CadQuery/cadquery #1506: Behaviour of ">>" face selector

bragostin, 2024-01-20:
I have spent quite some time in the documentation trying to understand why in the example below the face selection method 1 works as expected while method 2 yields an error message. For me both should select all faces of the extruded tube.

```python
import cadquery as cq

tube_length = 43.
tube_radius = 9. / 2.
tube_thickness = 1.5

tube = cq.Workplane("XY").circle(tube_radius).circle(tube_radius+tube_thickness).extrude(tube_length)

# FACE SELECTION METHOD 1
face = tube.faces('>>Y').vals()[0]
cq.exporters.export(face, "face_0.step")
face = tube.faces('>>Y').vals()[1]
cq.exporters.export(face, "face_1.step")
face = tube.faces('>>Y').vals()[2]
cq.exporters.export(face, "face_2.step")
face = tube.faces('>>Y').vals()[3]
cq.exporters.export(face, "face_3.step")

# FACE SELECTION METHOD 2
face = tube.faces('>>Y[0]').vals()[0]
cq.exporters.export(face, "face_Y0.step")
face = tube.faces('>>Y[1]').vals()[0]
cq.exporters.export(face, "face_Y1.step")
face = tube.faces('>>Y[2]').vals()[0]
cq.exporters.export(face, "face_Y2.step")
face = tube.faces('>>Y[3]').vals()[0]
cq.exporters.export(face, "face_Y3.step")

```

lorenzncode, 2024-01-21:
All of the faces have center of ~(0, 0) in x, y.

This 
```py
tube.faces('>>Y[1]')
```

is similar to (fails):

```py
tube.faces(cq.selectors.CenterNthSelector(cq.Vector(0, 1, 0), 1, True, 1e-4))
```

Changing the tolerance to an extremely small value will avoid the error but I don't think that is a reliable way to select the face.

I'd try something like this: 
```py
f0 = tube.faces("<Z")
f1 = tube.faces(">Z")
f2 = tube.faces("not |Z").faces(cq.selectors.AreaNthSelector(0))
f3 = tube.faces("not |Z").faces(cq.selectors.AreaNthSelector(1))
```

bragostin, 2024-01-21:
I understand, I was missing the proper use of `AreaNthSelector`.  Thank you for taking the time to explain!

closed 2024-01-21 by bragostin

lorenzncode, 2024-01-21:
No problem. Here is another way to select cylinder faces (say if you did not have explicit direction information).

```py
# two seam edges
eseams = tube.edges("%LINE")

# two faces
fcircle = eseams.ancestors("Face")

f2 = fcircle.faces(cq.selectors.AreaNthSelector(0))
f3 = fcircle.faces(cq.selectors.AreaNthSelector(1))

seamdir = f2.edges("%LINE").val().tangentAt(0)
f0 = tube.faces(cq.selectors.DirectionSelector(-seamdir))
f1 = tube.faces(cq.selectors.DirectionSelector(seamdir))
```

bragostin, 2024-01-21:
@lorenzncode these are elegant solutions, thank you!