CadQuery/cadquery #1512: ifcopenshell to caquery

SubChange, 2024-01-28:
ifcopenshell is a powerful tool to parse ifc files. In the `ifcopenshel.geom` [part](https://blenderbim.org/docs-python/ifcopenshell-python/geometry_processing.html), I saw that one can convert ifc file to OpenCASCADE BRep by

```python
settings = ifcopenshell.geom.settings()
settings.set(settings.USE_PYTHON_OPENCASCADE, True)
```
However, above implementation is based on [pythonocc-core](https://github.com/tpaviot/pythonocc-core), another python wrapper of OpenCASCADE. I want to know if there are any methods to convert ifc file to cadquery objects?

jmwright, 2024-01-28:
Potentially related issues [here](https://github.com/CadQuery/cadquery/issues/518#event-8775347704).

adam-urbanczyk, 2024-01-28:
So the actual question is: how do I convert from PythonOCC to OCP. There are two options:

1.  Serialize to brep and then deserialize (see #518)
2. If PythonOCC can provide the TopoDS_Shape address you can use `OCP.TopoDS.TopoDS_Shape._from_address`