CadQuery/cadquery #1515: Section view on compounds?

GaN-T, 2024-01-31:
Hello,

Is there a way to perform a section view on a compound? I want to hollow out the inside of the compound (a hidden cavity) and visualize its cross section.

I have an assembly object that I converted to a compound using the toCompound() method. I then performed a cut on it with another object positioned within this compound to create the cavity. This works, but how can I visualize it? The .section() method doesn't apply in this case. Any suggestions?

best,
GaNT

GaN-T, 2024-01-31:
Ach! I see I could just use shell() with a positive thickness !

GaN-T, 2024-01-31:
So the shell command works for what I am trying to do in principle, but not at the dimensions I want, I need a very thin layer wrapped around an object, much thinner than the original dimensions, but this doesn't work and I get an error (StdFail_NotDone: BRep_API: command not done). I dont get an error when the shell thickness is 1e-3, but it fails when i try to make the shell thickness smaller. How does one make really thin shells?

 See example below:

```
b =  cq.Workplane("front").box(5,5,2.5e-3)
b.faces("+Z").shell(0.0001, kind = "intersection" )

```

adam-urbanczyk, 2024-02-02:
There is a tol parameter in the Shape API. Can you try if changing it solves the issue?

GaN-T, 2024-02-03:
Yes this solves the problem, but I had to go into the source file of the Shape API and manually change it. It was not possible to pass it as an argument in the normal call to shell() as a workplane method, which raises a type error. Could this be included too in the future release or is there a particular reason why it is hard coded to 0.0001

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[6], line 2
      1 b =  cq.Workplane("front").box(5,5,2.5e-3)
----> 2 b.faces("+Z").shell(1e-4, kind = "intersection", tol = 1e-6 )

TypeError: Workplane.shell() got an unexpected keyword argument 'tol'
```

adam-urbanczyk, 2024-02-03:
That's a reasonable request. In the meantime you can modify default arguments at runtime (see `methodname.__defaults__`).

closed 2024-02-03 by GaN-T