CadQuery/cadquery #1520: CAD generation issues when sweeping a circular cross-section along a spline

vmaurin-pf, 2024-02-06:
Hi there,

Basically, I am generating a solid by sweeping a small circular cross section (5 mm) around a circle generated by a spline with 1000 points (1 m radius). The code generating this solid is attached below.
Interestingly enough, when I import the generated step file in solidworks, the solid is loading fine. However, when loading this solid in the Cadquery GUI or in FreeCAD, only the right part of the solid seems to be generated (see picture 1 and 2).

I have narrowed down this error to the use of the spline. When generating the circle with a polyline, the step object seems to be loaded correctly in Solidworks, FreeCAD and in the Cadquery GUI. Do you have any idea of what is going on with the spline object? Am I using too many points? Or is there a tangency argument that I need to specify in the spline object as well? 

Thanks for your help!

[image]
[image]

# Code: 
import numpy as np
import cadquery as cq
import os

your_path = "write your file path"

# parameters
radius = 1
number_of_points_spline = 1000
diameter = 5e-3

# generating a spline
thetas = np.linspace(0, 2*np.pi, number_of_points_spline)
x_coord = np.cos(thetas)*radius
y_coord = np.sin(thetas)*radius

list_xy_coord = []

for coord_number in range(len(x_coord)):
    list_xy_coord.append((x_coord[coord_number], y_coord[coord_number]))

spline = cq.Workplane("XZ").spline(list_xy_coord).close()
#spline = cq.Workplane("XZ").polyline(list_xy_coord).close()

# generating a circular cross section
circular_cross_section = cq.Workplane("XY").center(radius, radius).circle(diameter)

# sweeping the cross section along the spline
circular_solid = circular_cross_section.sweep(spline)

# saving as a step file
cq.exporters.export(circular_solid, os.path.join(your_path, "circular_solid.step"))

adam-urbanczyk, 2024-02-06:
That seems to be a tessellation/rendering issue. AFAICT the model is valid. In general it is probably not a good idea to interpolate 1000 points and if circle is what you need use circle - the whole point of NURBS is that you can represent circles.

vmaurin-pf, 2024-02-07:
Ok that makes sense. Thanks a lot for your answer!